stereolabs/zed-sdk #127: OpenCV not found in Camera Control, SVO playback and export

ZHOUXIAOYI95, 2019-02-12:
Hi guys,

1. In the Camera Control tutorial, there is a lack of a 'hpp.' file, how do you solve it?
2. In the SVO Recording tutorial, the recording file works perfect, but the CMake can't configure the playback and export  . are there any missing files? 

Thanks in advance for the help.
[image]

obraun-sl, 2019-02-13:
Those samples requires OpenCV to run. Cmake does not find OpenCV because the OpenCV version we have in /dependencies folder is built for VS2015 and you are using VS2017.

--> Best would be to install OpenCV for VS2017 and then choose the installation path for OpenCV config path.

ZHOUXIAOYI95, 2019-02-14:
> Those samples requires OpenCV to run. Cmake does not find OpenCV because the OpenCV version we have in /dependencies folder is built for VS2015 and you are using VS2017.
> 
> --> Best would be to install OpenCV for VS2017 and then choose the installation path for OpenCV config path.

Thanks for your advice. Now I have successfully installed OpenCV for VS 2017 and also added it to Path in environment variables. Then I try to configure the "SVO playback" again, but I still failed and the dialog shows below. Could you please tell me some more information?
[image]

obraun-sl, 2019-02-14:
Now you need to set the path of your new opencv in the OPENCV_CONFIG_PATH (just below OPENCV_FOUND) variable in Cmake. Right now it's still C:/Program Files(x86)/ZED SDK/dependencies/opencv3.1.0/

ZHOUXIAOYI95, 2019-02-14:
> Now you need to set the path of your new opencv in the OPENCV_CONFIG_PATH (just below OPENCV_FOUND) variable in Cmake. Right now it's still C:/Program Files(x86)/ZED SDK/dependencies/opencv3.1.0/

Hi there,

I download the opencv4.0.1, which file do I need to set to the path? Is the file in C:/Program Files(x86)/ZED SDK/dependencies?  if so, I can't find the opencv4.0.1 in the dependencies file.

obraun-sl, 2019-02-15:
in  OPENCV_CONFIG_PATH, you need to set the path where you *installed* opencv4.0.1. It won't be ZED SDK/dependencies/

closed 2022-04-26 by github-actions[bot]