stereolabs/zed-sdk #129: Error when trying to record HD2K or HD1080 on Zed Explorer

shehzaman, 2019-02-19:
Trying to record SVO on Windows 10, without CUDA at 2K or 1080p resolution. Console output shows:

```bash
C:\Program Files (x86)\ZED SDK\tools>"ZED Explorer.exe"
 ** [SVO] Hardware compression (NVENC) not available **
QWindowsWindow::setGeometry: Unable to set geometry 1406x658+432+284 on QWidgetWindow/'QWidgetClassWindow'. Resulting geometry:  1044x658+432+284 (frame: 0, 0, 0, 0, custom margin: 0, 0, 0, 0, minimum size: 0x0, maximum size: 16777215x16777215).
Error in Shader source code
Information written in shaderLog.txt
Wrong location, does this var exist : "texInput"? Is it used in the program? (May be the GLCompiler swapped it)
    Trying : texInput<-0
Error in Shader source code
Information written in shaderLog.txt
Wrong location, does this var exist : "texInput"? Is it used in the program? (May be the GLCompiler swapped it)
    Trying : texInput<-0
```

Anytime I try to change the output to HD2K or HD1080 I get one instance of the `Error in Shader source code...Trying : texInput<-0` error. 

Additionally, the output color seems off. Here is a picture

[image]

obraun-sl, 2019-02-20:
Hi,
What is your graphic card /unit ? It seems to be an incompatibility between OpenGL/GLSL version used in the software and the one available on your PC. Can you check that your graphic drivers are up to date ?
Note: This does not impact the SVO recording however, it's more a rendering issue.

obraun-sl, 2019-02-20:
opengl extensions viewer can be used to check the opengl version (http://realtech-vr.com/admin/glview)

shehzaman, 2019-02-20:
Hi there,

Graphics are inbuilt Intel HD graphics - they seem to be up to date. I'll check with OpenGL version with `glview` soon.

I am okay with images not rendering. Just that I am not getting HD2K SVO recording. 

ZED explorer does not open when I pass `-r HD2K` to record SVO. It just exits without any error.

obraun-sl, 2019-02-20:
OK I've just tested it on a Surface with Intel HD Graphics 520 without problems. the OpenGL core version is 4.5. For ZED Explorer it must be higher than 3.2 (since GLSL used inside is 150).
For HD2K : does the ZED is in USB3.0 ? You can check that the resolution printed at the bottom left of the view is indeed the one you asked (not in command line then).

shehzaman, 2019-02-20:
Okay success! Thank you for all your help. OpenGL version is 3.2 with compatibility mode - so the rendering does not work correctly. GLSL 150 is not available, I found this in shaderLogs logs:

```bash
Compilation log from inner string

ERROR: 0:1: '' :  version '150' is not supported

Strange chars found : ( char - line - col )
    ( # - 1 - 1)
    ( & - 9 - 72)
    ( & - 9 - 73)
    ( & - 9 - 84)
    ( & - 9 - 85)
    ( & - 9 - 96)
    ( & - 9 - 97)
```

Did not need GUI for any use. So I am not concerned about it now. 

SVO Recording: I was only allowed to record at VGA - 30fps resolution. Turns out my laptop has blue USB connectors but they do not support USB 3.0 (there was no `SS` next to it). Any resolution higher than VGA gives the following error:

```bash
C:\Program Files (x86)\ZED SDK\tools>"ZED Explorer.exe" -r HD720 --cpm 0 -o test.svo
Can not open the ZED, please connect it to another USB port.
``` 

Tried with `--cpm 0/1/2` (hardware encoding is not available).

I guess any resolution above VGA will need USB 3.0? I'll find another laptop in a few minutes and test with USB 3.0

shehzaman, 2019-02-26:
Alright, the few minutes took longer than expected. I was successful in installing ZED SDK and running ZED Explorer fro recording at HD2K resolution. 

The new computer running on Ubuntu18.04 with a USB 3.0 port works. OpenGL has a few issues with finding the right version on Ubuntu18.04. I had to export the correct version of OpenGL in `.bashrc` and ZED Explorer worked as expected:

```bash
export MESA_GL_VERSION_OVERRIDE=$(glxinfo | grep Max | awk '{print $NF}')
```

Which turned out to be `4.2`

Thank you for all your help and insight @obraun-sl

closed 2019-02-26 by shehzaman