stereolabs/zed-sdk #130: Zed SDK cuda 10.1 and ubuntu 16

pamarinp, 2019-02-27:
I have tried the install file version ZED_SDK_Linux_Ubuntu16_v2.3.3.run located in the ZED usb and they said I have cuda version 10.1 but it works only with 9.0, therefore it will not work with cuda.

I download the last version ZED_SDK_Linux_Ubuntu16_v2.7.1.run and they said again I have installed cuda 10.1 but it required cuda 10.0 and zed will not work with cuda.

Do you know any install file that works with cuda 10.1 and ubuntu 16.04?

Hubert51, 2019-03-19:
Same question. It seems the zed does not work with cuda 10.1. Do you find some solutions?

agirgen, 2019-07-30:
+1, especially since the download link on the SDK page takes you to the 10.1 download.  And the Nvidia download page says its version 10, not version 10.1.  Time wasted.

snakehaihai, 2019-10-30:
+1. ZED need Cuda 10. So I choose Cuda 10 on nvidia download section. but nvidia gives me 10.1. So any hot fixs?

matlabbe, 2019-10-30:
Download 10.0 archive here: https://developer.nvidia.com/cuda-10.0-download-archive

closed 2022-04-26 by github-actions[bot]