stereolabs/zed-sdk #131: ZED SDK on XAVIER ISSUE

kdhakal18, 2019-03-01:
I am currenlty playing with Jetson Xavier Dev Kit. The sdk 2.7 for xavier is installed (the terminal says its installed at least). However, the tools in the tools directory are installed as  a shared library rather than executable. The tools cannot be accessed and the SDK doesnot recognise the camera. I tried to make it executable but did not succed. I have been using ZED sdk for my PC with ubuntu 16.04 LTS which works perfectly fine with all the APIs. But the current jetpack comes with 18.04 LTS image which I might not be familiar with. Any suggestions on this.

Myzhar, 2019-03-01:
Hi @kdhakal18,
I think you have installed the wrong version of the SDK. Have you download the SDK for the Xavier?
https://download.stereolabs.com/zedsdk/2.7/tegraxavier

kdhakal18, 2019-03-01:
Hi @Myzhar No, I have installed the right SDK. I went through some ubuntu forums and there were issues regarding the handling of executable files for 18.04 LTS. I will try to run update and upgrade ubuntu and reinstall SDK. Lets see if things get sorted out.

kdhakal18, 2019-03-01:
@Myzhar No success!!

Myzhar, 2019-03-01:
I forward the issue to the dev team and we will be back to you as soon as possible

kdhakal18, 2019-03-04:
.It works with giving the permission through terminal with "chmod" command. It makes sense as ubuntu 18.04 is based upon GNOME unlike 16.04 which was based on Unity. Thanks for the support

closed 2022-04-26 by github-actions[bot]