stereolabs/zed-sdk #132: Greenscreen example not working

kvskranthikumar, 2019-03-07:
When we have used your green screen example, the scene in the ZED camera display is also moving. But the video in the example shows as-if the ZED camera display is static. 

https://www.stereolabs.com/blog/how-to-make-mixed-reality-vr-videos-in-unity/6/

Are they any additional settings to make the ZED camera display static (should not move in sync with HTC Vive headset). 

We want only the scene from headset mix with ZED camera output and not the position tracking. Attached recording might explain the situation better. When I’m moving my head, an environment is also moving. 

[IMG_3894.zip](https://github.com/stereolabs/zed-examples/files/2941634/IMG_3894.zip)

Need your help to come out of a situation. Appreciate your support.

ckorris, 2019-03-07:
I just opened up the sample with a Vive on, ran it, and did not have the same issue as you. 

Just to be clear, this happens in the greenscreen demo scene without any modifications? 

Also, in the Scene view, can you see the camera object also moving when the headset moves, or is it just the image? 

If you've made any changes, or imported any plugins (like SteamVR) could you let me know, and ideally post an image of the hierarchy? 

Often this can happen if the ZED rig is somehow parented to the VR camera.

kvskranthikumar, 2019-03-08:
Thanks, Chris.

Yes, we have used the GreenScreen demo without any modification to project. We got the Zed plugin into Unity and trying to run it.

The whole scene view is moving as we move the headset. You can see this in the earlier video attachment shared.

Yes, we have imported SteamVR and a small change to the Clear Flags as "Depth only" otherwise our screen is showing the sky color as blue. The modification is highlighted in the image attached.

[image]

Please let us know if you need any more detail.

ckorris, 2019-03-08:
Okay, I switched that camera to Depth Only, and in the _Game_ window I can see similar results as you - in the background, you have the appearance of the camera moving because the image of the last camera to render (the VR camera) isn't cleared before it renders. So all you have to do is calibrate or position the ZED camera in the scene so it's not seeing anything (like under the world, up at the sky, etc.) and my video looks quite close to yours. If this were your issue, switching back to Skybox for the Clear Flags would solve it. 

But what's stumping me is that you said that the ZED camera is also moving in the _Scene_ view in addition to the Game view, which in my case doesn't happen at all, and I can't think of why it would. See video: https://youtu.be/3DRzflPz3mY

Are you positive the ZED Camera is moving in the Scene view as I asked? Because if so, I'm not sure where to start. Does it also occur if you open a project without SteamVR imported?

closed 2022-04-26 by github-actions[bot]