stereolabs/zed-sdk #134: Build depth sensing example in Qt

morimkb, 2019-03-12:
Hi,
I am trying to build the depth sensing example in Qt qmake, but I have issues related to INCLUDE libraries such as Zed and Cuda. I have written INCLUDEPATH += "/usr/local/zed/include/" \ as my Zed library path in .pro file but I still get many errors like this "error: undefined reference to `sl::Camera::Camera()'"

Can someone tell how I can create a qmake file for zed library?

obraun-sl, 2019-03-13:
Hi,

In your pro file you should have : 
unix:INCLUDEPATH  += "/usr/local/zed/include"
unix:LIBS += -L"/usr/local/zed/lib"
unix:LIBS += -lsl_input
unix:LIBS += -lsl_core
unix:LIBS += -lsl_zed

Depending on your cuda version, you also need to add includepath and libs/lib dirs for CUDA : For example : 
unix:INCLUDEPATH  += "/usr/local/cuda-10.0/include"
unix:LIBS += -L"/usr/local/cuda/lib64"
unix:LIBS += -lcuda
unix:LIBS += -lcudart
unix:LIBS += -lnppicom
unix:LIBS += -lnppial
unix:LIBS += -lnppidei
unix:LIBS += -lnppif
unix:LIBS += -lnppig
unix:LIBS += -lnppim
unix:LIBS += -lnppist
unix:LIBS += -lnppisu
unix:LIBS += -lnppitc
unix:LIBS += -lnppc
unix:LIBS += -lnpps

Then in you header file, you just need to include 
#include <sl/camera.hpp>

Best,
OB

XuErShuo, 2022-04-23:
> 你好，
> 
> 在您的专业文件中，您应该有：
> 
> ```
> unix:INCLUDEPATH  += "/usr/local/zed/include"
> unix:LIBS += -L"/usr/local/zed/lib"
> unix:LIBS += -lsl_input
> unix:LIBS += -lsl_core
> unix:LIBS += -lsl_zed
> ```
> 
> 根据您的 cuda 版本，您还需要为 CUDA 添加 includepath 和 libs/lib 目录：例如：
> 
> ```
> unix:INCLUDEPATH  += "/usr/local/cuda-10.0/include"
> unix:LIBS += -L"/usr/local/cuda/lib64"
> unix:LIBS += -lcuda
> unix:LIBS += -lcudart
> unix:LIBS += -lnppicom
> unix:LIBS += -lnppial
> unix:LIBS += -lnppidei
> unix:LIBS += -lnppif
> unix:LIBS += -lnppig
> unix:LIBS += -lnppim
> unix:LIBS += -lnppist
> unix:LIBS += -lnppisu
> unix:LIBS += -lnppitc
> unix:LIBS += -lnppc
> unix:LIBS += -lnpps
> ```
> 
> 然后在你的头文件中，你只需要包含
> 
> ```
> #include <sl/camera.hpp>
> ```
> 
> 最好的， OB
The Latest file is #include <sl/Camera.hpp>，which should be capital.

closed 2022-05-29 by github-actions[bot]