stereolabs/zed-sdk #135: Repair the depth image by the RGBA image

hanchengzhe, 2019-03-16:
[image]
Hi, I am working on the depth image returned by ZED. I found the depth information of some points is not accurate(such as the area around my finger), how can this be fixed by the RGBA image or normal image using color and texture features?

WASCHMASCHINE, 2019-03-18:
You could turn off the fill option, which is more accurate but less dense with holes. You could then try to separate background and foreground pixels via color alone if your goal is hand-tracking. Then you could interpolate the background depth around the hand contours.

Alternatively do your own stereo by training a CNN on many similar images like this with ground truth.

GiuseppeLozioEllena, 2019-06-25:
Hi, I have a similar problem with Unity Zed plugin. It's possible to increase the precision of occlusion on holes. I want to avoid this :
[image]

closed 2022-04-26 by github-actions[bot]