stereolabs/zed-sdk #138: ZED camera error - Camera already in use

DaniSubodh, 2019-03-23:
Hello,
We are trying to connect ZED stereo cam with our new Ubuntu PC (Please note that it works fine with our previous machine).
The camera is detected, but we end up with an error in Camera test (ZED Diagnostic):
**Camera already in use**
Several program cannot use ZED at the same time, make sure to close them.

[image]

The config currently used is:
Ubuntu - 4.15.0-46-generic #49~16.04.1-Ubuntu SMP  x86_64 x86_64 x86_64 GNU/Linux
CUDA Toolkit version: "V10.0.130"
ZED SDK - ZED_SDK_Ubuntu16_v2.7.1

Also please find the Diagnostic results file attached:
[ZED_Diagonistic_Results.txt](https://github.com/stereolabs/zed-examples/files/2999393/ZED_Diagonistic_Results.txt)

Thank you,
Subodh

josiasalexandre, 2019-05-06:
Hi,
I have the same problem over here.
Ubuntu - 4.15.0.48-generic -- 16.04
CUDA Tookit version: 9.0
ZED SDK - 2.8

The camera works only the first time.

Context:

1. Connect the camera
2. Open the ZED Explorer
3. The camera works fine and I can see the video
4. Close the ZED Explorer
5. Open the ZED Explorer again or any other program which open the camera.
6. The camera doesn't work anymore and the diagnostic tool indicates that the camera is already in use and Several program cannot use the ZED...

I'm using the latest available firmware.

jarvis-huang, 2019-05-24:
Same issue for me. Ubuntu 16 + CUDA 9.
No program is using the camera but ZED Diagnostic still says 'camera in use'.

epoc88, 2019-05-27:
Same issue too. Ubuntu 16.04 CUDA 10.0, ZED SDK 2.8

The ZED explorer is frozen even if no camera is running.

MoawyaWahdan, 2019-06-12:
Running **multiple cameras**  in ubuntu 16 cuda 8  with any linux kernel newer than 4.9  causes problems and works only with kernal 4.9.70. I thought the issue will be fixed when upgrading the system to ubuntu 18, kernal 4.18.0 and cuda 10. And upgrading zed sdk from 2.2.0 to 2.8.1 and firmware of cameras to latest. **Unfortunately  error "process is already in use" still happening when connecting 3 cameras and i had to revert back to kernal 4.9.70 to fix it. Any explanation?**

cm107, 2019-12-19:
After you open the zed camera, like say in python, you have to close it before you can open it again.
In my case, I was running into errors from the start so I didn't even notice, but my python script did indeed leave the camera open before throwing an error.
Doing the Diagnostic test afterwards showed that the camera was busy, but when I tried opening the ZED Explorer once and then closed it, everything seemed to be reset properly and working again.

closed 2022-04-26 by github-actions[bot]