stereolabs/zed-sdk #139: make error -path issue

AndreV84, 2019-03-25:
any solution?
 make
Scanning dependencies of target ZED_with_OpenCV
[ 33%] Building CXX object CMakeFiles/ZED_with_OpenCV.dir/src/SaveDepth.o
[ 66%] Building CXX object CMakeFiles/ZED_with_OpenCV.dir/src/main.o
[100%] Linking CXX executable ZED_with_OpenCV
/usr/bin/x86_64-linux-gnu-ld: warning: libnppial.so.10.0, needed by /usr/local/zed/lib/libsl_core.so, not found (try using -rpath or -rpath-link)
/usr/bin/x86_64-linux-gnu-ld: warning: libnppisu.so.10.0, needed by /usr/local/zed/lib/libsl_core.so, not found (try using -rpath or -rpath-link)
/usr/bin/x86_64-linux-gnu-ld: warning: libnppidei.so.10.0, needed by /usr/local/zed/lib/libsl_core.so, not found (try using -rpath or -rpath-link)
/usr/bin/x86_64-linux-gnu-ld: warning: libnppif.so.10.0, needed by /usr/local/zed/lib/libsl_core.so, not found (try using -rpath or -rpath-link)
/usr/bin/x86_64-linux-gnu-ld: warning: libnppist.so.10.0, needed by /usr/local/zed/lib/libsl_zed.so, not found (try using -rpath or -rpath-link)
/usr/bin/x86_64-linux-gnu-ld: warning: libnpps.so.10.0, needed by /usr/local/zed/lib/libsl_zed.so, not found (try using -rpath or -rpath-link)
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppsMalloc_32f@libnpps.so.10.0'
/usr/local/zed/lib/libsl_core.so: undefined reference to `nppiFilterBoxBorder_8u_C4R@libnppif.so.10.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppiMulC_32f_C1IR@libnppial.so.10.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppsMalloc_32s@libnpps.so.10.0'
/usr/local/zed/lib/libsl_core.so: undefined reference to `nppiAdd_32f_C1R@libnppial.so.10.0'
/usr/local/zed/lib/libsl_core.so: undefined reference to `nppiSet_32f_C1R@libnppidei.so.10.0'
/usr/local/zed/lib/libsl_core.so: undefined reference to `nppiMalloc_32f_C1@libnppisu.so.10.0'
/usr/local/zed/lib/libsl_core.so: undefined reference to `nppSetStream'
/usr/local/zed/lib/libsl_core.so: undefined reference to `nppGetStream'
/usr/local/zed/lib/libsl_core.so: undefined reference to `nppiFree@libnppisu.so.10.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppsMalloc_8u@libnpps.so.10.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppiHistogramRangeGetBufferSize_32f_C1R@libnppist.so.10.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppiFilterBoxBorder_8u_C1R@libnppif.so.10.0'
/usr/local/zed/lib/libsl_core.so: undefined reference to `nppiMalloc_8u_C1@libnppisu.so.10.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppiHistogramRange_32f_C1R@libnppist.so.10.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppsFree@libnpps.so.10.0'
collect2: error: ld returned 1 exit status
CMakeFiles/ZED_with_OpenCV.dir/build.make:131: recipe for target 'ZED_with_OpenCV' failed
make[2]: *** [ZED_with_OpenCV] Error 1
CMakeFiles/Makefile2:72: recipe for target 'CMakeFiles/ZED_with_OpenCV.dir/all' failed
make[1]: *** [CMakeFiles/ZED_with_OpenCV.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

I have the files, as it seems to me, but what is the proper way to resolve the path issue?
cmake -rpath /usr/local/cuda-10.1/targets/x86_64-linux/lib/libnppial.so.10 ..
wont work

AndreV84, 2019-03-25:
usr/local/zed/sample/svo recording/recording/build$ cmake -rpath /usr/local/cuda-10.1/targets/x86_64-linux/lib/libnppial.so.10 ..
results in the same error:
make
[ 50%] Linking CXX executable ZED_SVO_Recording
/usr/bin/x86_64-linux-gnu-ld: warning: libnppial.so.10.0, needed by /usr/local/zed/lib/libsl_core.so, not found (try using -rpath or -rpath-link)
/usr/bin/x86_64-linux-gnu-ld: warning: libnppisu.so.10.0, needed by /usr/local/zed/lib/libsl_core.so, not found (try using -rpath or -rpath-link)
/usr/bin/x86_64-linux-gnu-ld: warning: libnppidei.so.10.0, needed by /usr/local/zed/lib/libsl_core.so, not found (try using -rpath or -rpath-link)
/usr/bin/x86_64-linux-gnu-ld: warning: libnppif.so.10.0, needed by /usr/local/zed/lib/libsl_core.so, not found (try using -rpath or -rpath-link)
/usr/bin/x86_64-linux-gnu-ld: warning: libnppist.so.10.0, needed by /usr/local/zed/lib/libsl_zed.so, not found (try using -rpath or -rpath-link)
/usr/bin/x86_64-linux-gnu-ld: warning: libnpps.so.10.0, needed by /usr/local/zed/lib/libsl_zed.so, not found (try using -rpath or -rpath-link)
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppsFree@libnpps.so.10.0'
/usr/local/zed/lib/libsl_core.so: undefined reference to `nppiMalloc_8u_C1@libnppisu.so.10.0'
/usr/local/zed/lib/libsl_core.so: undefined reference to `nppiFree@libnppisu.so.10.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppiHistogramRange_32f_C1R@libnppist.so.10.0'
/usr/local/zed/lib/libsl_core.so: undefined reference to `nppiSet_32f_C1R@libnppidei.so.10.0'
/usr/local/zed/lib/libsl_core.so: undefined reference to `nppiAdd_32f_C1R@libnppial.so.10.0'
/usr/local/zed/lib/libsl_core.so: undefined reference to `nppSetStream'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppiHistogramRangeGetBufferSize_32f_C1R@libnppist.so.10.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppsMalloc_32f@libnpps.so.10.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppsMalloc_8u@libnpps.so.10.0'
/usr/local/zed/lib/libsl_core.so: undefined reference to `nppGetStream'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppiFilterBoxBorder_8u_C1R@libnppif.so.10.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppiMulC_32f_C1IR@libnppial.so.10.0'
/usr/local/zed/lib/libsl_core.so: undefined reference to `nppiMalloc_32f_C1@libnppisu.so.10.0'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `nppsMalloc_32s@libnpps.so.10.0'
/usr/local/zed/lib/libsl_core.so: undefined reference to `nppiFilterBoxBorder_8u_C4R@libnppif.so.10.0'
collect2: error: ld returned 1 exit status
CMakeFiles/ZED_SVO_Recording.dir/build.make:99: recipe for target 'ZED_SVO_Recording' failed
make[2]: *** [ZED_SVO_Recording] Error 1
CMakeFiles/Makefile2:72: recipe for target 'CMakeFiles/ZED_SVO_Recording.dir/all' failed
make[1]: *** [CMakeFiles/ZED_SVO_Recording.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

obraun-sl, 2019-03-25:
Hi,
You need to install CUDA 10.0 and not CUDA 10.1 as we are not compatible with 10.1.
Best,
OB

AndreV84, 2019-03-25:
thank you for letting me know!
and is there a way to automatically build the samples, especially the svo record sample while installing zed sdk on cuda 10.0 computer in a way it wont require manual execution of make command afterwards?

obraun-sl, 2019-03-25:
You can do a script that install the SDK and *then* build the sample : 
something like : 

<pseudo code>
```
./ZED_SDK.....run --silent
cd /usr/local/zed/samples
cd svo\ recording
mkdir build
cd build
cmake .. && make -j2
```

AndreV84, 2019-03-25:
thank you for letting me know!

AndreV84, 2019-03-26:
rather:
cd /usr/local/zed/sample
cd svo\ recording
 cd recording
mkdir build
cd build
cmake .. && make

AndreV84, 2019-03-27:
and python? cpp works, but python?
 python3 zed-opencv.py ../cpp/build/VGA_SN2757_13-05-55.svo 
Traceback (most recent call last):
  File "zed-opencv.py", line 3, in <module>
    import pyzed.sl as sl
ModuleNotFoundError: No module named 'pyzed'

AndreV84, 2019-03-27:
it seesm zed-python-ai is required
https://github.com/stereolabs/zed-python-api

closed 2019-03-27 by AndreV84