stereolabs/zed-sdk #141: Problem running

UannaFF, 2019-03-29:
Hello, i'm running into a problem when i try to run almost every program:

./ZED Depth Viewer: error while loading shared libraries: libnppc.so.10.0: cannot open shared object file: No such file or directory

I checked with locate and I saw i have the file installed:

/usr/local/cuda-10.0/doc/man/man7/libnppc.so.7
/usr/local/cuda-10.0/lib64/libnppc.so
/usr/local/cuda-10.0/lib64/libnppc.so.10.0
/usr/local/cuda-10.0/lib64/libnppc.so.10.0.130
/usr/local/cuda-10.0/lib64/stubs/libnppc.so
/usr/local/cuda-9.1/doc/man/man7/libnppc.so.7
/usr/local/cuda-9.1/lib64/libnppc.so
/usr/local/cuda-9.1/lib64/libnppc.so.9.1
/usr/local/cuda-9.1/lib64/libnppc.so.9.1.85
/usr/local/cuda-9.1/lib64/stubs/libnppc.so
/usr/share/man/man7/libnppc.so.7

LoicDrd, 2019-05-13:
Hello, I have the same issue. Have you found a solution?

irenesanznieto, 2019-07-08:
Hi, I'm not sure if you still have this issue. I had the same error because I didn't follow the last part of the CUDA's installation output:
You have to include /usr/local/cuda-10.0/bin to your PATH env variable and /usr/local/cuda-10.0/lib64 to your LD_LIBRARY_PATH env variable. 

I hope this helps you.

GentleDell, 2019-10-09:
Hi, I have the same issue when I try to execute ZED samples in Qt. 

For example, when I run the ZED_Camera_Control, Qt raises an error: 
"ZED_Camera_Control: error while loading shared libraries: libnppc.so.10.0: cannot open shared object file: No such file or directory."

But when I compile and run ZED_Camera_Control in terminal, there is no error. What's more interesting is, when I run ZED_Camera_Control (generated by Qt) in terminal, there is no error. However, if I set Qt to run in terminal, the error still exists. 

So, in summary: 
compile in Qt + run in Qt = Error
compile in Qt + set "run in terminal" in Qt = Error
compile in Qt + open a terminal and run the sample = Success
compile in terminal + run in terminal = Success

I was wondering whether there was any solution? 

Thank you very much for your time.

obraun-sl, 2019-10-10:
Hi,
I suppose you are talking about QtCreator for the IDE. 
You need to add the cuda/lib64 path in LD_LIBRARY_PATH in the run environment in Projects/Run settings.

GentleDell, 2019-10-10:
@obraun-sl ,
Hi,

Thank you so much. It works now.  
Thanks again.

Best,

closed 2019-10-10 by obraun-sl