stereolabs/zed-sdk #142: 2.7.1 bug: delay when recording multiple .svo files

amaanda, 2019-04-12:
Me and my team have been using ZED for our application and it worked fine with version 2.4.0. Since we upgraded for 2.7.1, we have been having problems when recording multiple .svos continuously: from one recording to the next one, the SDK have a delay of up to **40 seconds** from one file to the next one. We ran multiples tests with 2 different situations: (1) we recorded a .svo file, stopped the recording (saving the .svo), and then waited until the camera preview became "live" again, then we hit the record button again. (2) we recorded a .svo fil,e stopped the recording, and did not wait for the camera preview to be "live" again, we just hit the record button right after stopping the previous recording.

I am very confused about what may be causing the problem. I assume it's a bug of the SDK because it did not happen in any of the previous versions we used.

obraun-sl, 2019-04-12:
Hi,
The SVO recording behavior has changed between 2.4 et 2.7 : 
Previously in 2.4, the SVO recording was an encoding followed by the data writting directly in the SVO file. 
Now, the Encoding and the data writting are running into 2 separated thread, so that the encoding fills a queue to be written into the file by another thread. This makes the SVO recording more efficient (it drops less frames than before), but if the disk speed is the limit,  when the SVO recording is disable and if there is some data not written in the file, the function will block until all the data are written in the file.

I would recommend to use the hardware-based encoder in 2.7 (SVO_COMPRESSION_AVCHD or SVO_COMPRESSION_HEVC) since the amount of data written in the SVO file is much much less, and the disk speed cannot be the limitation then. Note that it requires a NVIDIA GPU for now.
Otherwise, always try to record on a SSD to get the maximum disk speed, or decrease the capture framerate.

Best,
OB.

amaanda, 2019-04-12:
Is there a way to get the 2.7.1 to start recording without finishing the writing thread?

obraun-sl, 2019-04-12:
No really since the enableRecording() will wait for the disable to finish anyway. And the disableRecording also creates an indexing table at the end, therefore the thread cannot be killed externally.

Regarding what you want to do, I would really use the AVCHD or HEVC compression mode, since you won't have this long finish time.

amaanda, 2019-04-12:
Not even if a start a new thread to start recording while the previous one finishes writing?

obraun-sl, 2019-04-12:
I don't think the thread will work, or to protect the ressources, you will need to use mutex which will lead to the same behavior at the end.

The AVCHD compression should already work with the GT940MX (but not the HEVC) since it's compatible with all the NVIDIA graphic card supported by the SDK.

amaanda, 2019-04-12:
AVCHD won't work for our application since we lose depth information.

obraun-sl, 2019-04-12:
I am not sure to understand what do you mean by losing depth information ?

amaanda, 2019-04-12:
I am sorry I was not very clear. I meant that we lose information from the image itself, therefore we are gettin less information for the stereo algorithm we use.

obraun-sl, 2019-04-12:
Ok I get it. Then if you want to have the exact original image in bits, then only LOSSLESS compression can be used.
Since on lossless, the amount of data to record is quite huge, you can try my second suggestion : 
"record on a SSD to get the maximum disk speed, or decrease the capture framerate (if no in 2K), or not call record at every grab()."

obraun-sl, 2019-05-10:
The SDK 2.8 was not focused on that, but mostly on new streaming feature and AVCHD/HEVC codecs. The recording behavior should be the same with 2.8.

closed 2022-04-26 by github-actions[bot]