stereolabs/zed-sdk #146: Can someone give the cuda version vs SDK match list.

Marxlp, 2019-05-20:
I cannot  find any info about which version of cuda can I used according to the version of the current sdk. I have old laptop equipped with GeForce GT 630M, which I use to test a ZED camera. CUDA 10.1 needs newest Driver version v425 for graphic card. However the dirver version of GT 630M currently only update to v391. I want to try CUDA 9.1, But I am not sure whether it would work. Does anyone know it?

WASCHMASCHINE, 2019-05-20:
Check the release notes: 2.3 is the first version to support CUDA 9.1: 
https://www.stereolabs.com/developers/release/2.3/

Marxlp, 2019-05-24:
Thanks, I try CUDA 9.0 which demanding lower version of Nvidia graphics card. It worked but failed running the depth viewer due to limited memory of the graphic card. Anyway, thanks for your help.

closed 2019-05-24 by Marxlp