stereolabs/zed-sdk #147: Can't convert SVO files into (SEQUENCE LEFT+DEPTH)

FabianAC07, 2019-05-23:
Hello, 

I am trying to convert my svo files which I recorded using the ZED explorer to sequences of images on LEFT RGB + Depth but I get the following on cmd:

```
`C:\Program Files (x86)\ZED SDK\samples\bin> ZED_SVO_Export "C:\Users\fabia\Documents\ZED\HD720_SN10027514_15-55-57.svo\" "C:\Users\fabia\Documents\ZED\image\" 3
Usage:

    ZED_SVO_Export A B C

Please use the following parameters from the command line:
 A - SVO file path (input) : "path/to/file.svo"
 B - AVI file path (output) or image sequence folder(output) : "path/to/output/file.avi" or "path/to/output/folder"
 C - Export mode:  0=Export LEFT+RIGHT AVI.
                   1=Export LEFT+DEPTH_VIEW AVI.
                   2=Export LEFT+RIGHT image sequence.
                   3=Export LEFT+DEPTH_VIEW image sequence.
                   4=Export LEFT+DEPTH_16Bit image sequence.
 A and B need to end with '/' or '\'

Examples:
  (AVI LEFT+RIGHT)   ZED_SVO_Export "path/to/file.svo" "path/to/output/file.avi" 0
  (AVI LEFT+DEPTH)   ZED_SVO_Export "path/to/file.svo" "path/to/output/file.avi" 1
  (SEQUENCE LEFT+RIGHT)   ZED_SVO_Export "path/to/file.svo" "path/to/output/folder" 2
  (SEQUENCE LEFT+DEPTH)   ZED_SVO_Export "path/to/file.svo" "path/to/output/folder" 3
  (SEQUENCE LEFT+DEPTH_16Bit)   ZED_SVO_Export "path/to/file.svo" "path/to/output/folder" 4

Press [Enter] to continue`
```
Options 0  and 1 work fine, but image sequence options non of them work.

I am using the lates ZED STK v2.8.0

Can someone please help me solving this issue?

[image]

WASCHMASCHINE, 2019-05-23:
Hey Fabian, you can debug it yourself if you want. The SVO Exporter is included in the samples folder of the ZED SDK. You just need to build it.

FabianAC07, 2019-05-23:
> Hey Fabian, you can debug it yourself if you want. The SVO Exporter is included in the samples folder of the ZED SDK. You just need to build it.

Hello @WASCHMASCHINE I have tried to debug it but I am not an expert on this, so no luck so far, can you please point me in the right direction?

I have followed the instructions that Stereolabs provide to build the source code in the following link:

- [https://github.com/stereolabs/zed-examples/tree/master/svo%20recording/export](url)

But I get the following error on Cmake-gui: 

`CMake Error: Cannot open file for write: C:/Program Files (x86)/ZED SDK/samples/svo recording/export/build/CMakeCache.txt.tmp
CMake Error: The source directory "C:/Program Files (x86)/ZED SDK/samples/svo recording/export/src" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
CMake Error: : System Error: No such file or directory
CMake Error: Unable to open cache file for save. C:/Program Files (x86)/ZED SDK/samples/svo recording/export/build/CMakeCache.txt
CMake Error: : System Error: No such file or directory`

Even though the files are in the directory...

[image]

FabianAC07, 2019-05-23:
I have found the solution... 

It is not clear on the documentation but here it goes if anyone else have the same problem in the future:

Follow the next steps:

1. Run cmd.
2. Direct to the bin folder `cd C:\Program Files (x86)\ZED SDK\samples\bin`
3. Type the following arguments per instructions in my initial post

- ZED_SVO_Export A B C`

- A - SVO file path (input) : "path/to/file.svo"

 

- B - AVI file path (output) or image sequence folder(output) : "path/to/output/file.avi" or "path/to/output/folder"

-  C - Export mode: 0, 1, 2, 3 or 4

`ZED_SVO_Export "path\to\file.svo" "path\to\output\folder/" 3` 

- NOTE that at the end of the output path folder it should be followed by a forward slash `/`

closed 2019-05-23 by FabianAC07