stereolabs/zed-sdk #148: video choppy from Jetson TX2 to Windows and unexpected size pack

pinjiang, 2019-05-25:
Streaming Video from JETSON TX2 to a Windows PC, the video is  choppy and the console prints consecutive lines :

(ZED Stream) Received unexpected size pack 144/65016 ( ip: 193.168.100.249 )
(ZED Stream) Received unexpected size pack 144/65016 ( ip: 193.168.100.249 )
(ZED Stream) Received unexpected size pack 144/65016 ( ip: 193.168.100.249 )

I have verified that the sender MTU is smaller than the receiver end.

obraun-sl, 2019-05-27:
Hi,

The messages means that an amount of packet are lost during the client - server transmission.

To reduce the amount of packet loss : 
- Try to reduce the bitrate and increase the gop size (up to 256) (in the streaming parameters)
- Make sure that the Jetson is setup to maximum performances, especially for the Network settings. Use nvpmodel -m 0 or jetson clocks to activate max perf.
- Check your network configuration. Try to test on a smaller network configuration with just the Jetson and the Windows PC on it to avoid collisions.

Best,
OB

wkoa, 2019-06-27:
Hi, I am also trying to transport video from TX2 to Win10, but I was failed.
In Win10(Receiver), the program displays "ERROR_CODE_CAMERA_NOT_DETECTED".
So can you tell me some information about How to transport video from TX2 to Win10?

P.s 我想把ZED的视频从TX2传到Win10上去，但是在Win10上，也就是VS2015一直就显示 无法检测到摄像头，用的直接就是官方的环境。。。想请问你是怎么完成视频的传输的？你的配置环境是什么?
我现在的配置环境是：
TX2 和 Win10 都是CUDA9 和 SDK2.8.2，相机Firmware 都是1142

pinjiang, 2019-07-03:
ZED provides a sample application which enable you do streaming.
You should not meet ERROR_CODE_CAMERA_NOT_DETECTED problem since you don't need to have ZED at receiver side.

wkoa, 2019-07-04:
OK, Fine. thx

BingXiong1995, 2020-08-13:
After I change to h.256, the problem was solved.

closed 2022-04-26 by github-actions[bot]