stereolabs/zed-sdk #149: ZED Explorer not opening

AmbarishGK, 2019-05-27:
I am stuck not able to run the zed explorer with jetson nano developer kit. I have successfully installed the ZED SDK manager for it but I keep getting this error. 
[image]

So how do I open the explorer and start working with depth caemera?

nesnes, 2019-05-27:
Hi, it seems that the issue comes from a typo in your command.
`./ZED\Explorer` should be `./ZED\ Explorer` (with a space after the `\`)

AmbarishGK, 2019-05-27:
Heyy @nesnes , thanks for that. 
I solved it by making changing the filename to ZED_Explorer and the file executable using
`chmod +x  ZED_Explorer`
`./ZED_Explorer`
It worked!!

closed 2019-05-27 by AmbarishGK