stereolabs/zed-sdk #150: Error: ZED (Init) >> ERROR: Executable was designed for SDK v2.8.0 / installed v2.8.1

Docker300, 2019-06-03:
I have Visual Studio 2015 and also VS 2017, CUDA 10.0, cuDNN 7.5, OpenCV 3.4.0, CMAKE 3.13.4
I was able to run successfully all ZED Master examples
but I don't know why I am suddenly getting this error

**> ZED (Init) >> ERROR: Executable was designed for SDK v2.8.0 / installed v2.8.1
CAMERA FAILED TO SETUP**

I downloaded the SDK again and run the examples both in VS 2015/ 2017 after new installation of SDK. But could not fix the error

awm-pezhilarasan, 2019-06-11:
Hi, I'm facing the same problem in ubuntu. Is it possible to send link of 2.8.0 legacy releases? I was only able to find 2.7 other than 2.8.1

Thank you.

padkeemao, 2019-06-11:
Hello, I'm facing the same issue in ubuntu. It'd be great if you could attach the link to the 2.8.0 release. I can only find 2.7 release other than 2.8.1

Thank you.

Docker300, 2019-06-12:
It worked for me 
https://github.com/stereolabs/zed-yolo/issues/10

[image]

closed 2022-04-26 by github-actions[bot]