stereolabs/zed-sdk #151: sl::Camera::close() invalidates CUDA device memory?

fishcu, 2019-06-07:
I'm running into an issue where `sl::Camera::close()` causes ZED-unrelated CUDA code to fail with a "invalid device pointer" error.
This happens upon program shutdown where I first close the sl::Camera object, then another object's destructor tries to free memory. The error goes away if I comment out the `sl::Camera::close()` call.

- The unrelated CUDA code uses the default CUDA stream. Is it possible that the ZED SDK uses the default stream and somehow affects it when calling `sl::Camera::close()`?

I'm on Ubuntu 16.04LTS, CUDA 10.0, nvidia driver 418.67.

obraun-sl, 2019-06-10:
Hi,
`sl::Camera::close()` destroys the CUDA context if the context was created by the ZED SDK (i.e. without specify your own context in `sl::InitParameters`). Therefore, if some gpu pointer are called after close was called they will be invalid since the context does not exist anymore.

You have 2 solutions : 
- Call `close()` only when all your cuda memories are already freed. 
- Create your own cuda context (`cuCtxCreate()`) and pass the CUContext to `sl::InitParameters::sdk_cuda_ctx`. The context will not be destroyed when close() is called but you have to take care then about the context destruction in your code.

fishcu, 2019-06-11:
Thank you, that clarifies things. Is there a way to make the ZED SDK use the primary (default) context? Or what would be the best way to avoid (expensive) context switches and duplicate resource usage?

fishcu, 2019-06-11:
I solved it in the following way now. Does that seem plausible to you?
```CUcontext primary_cuda_context;
cuCtxGetCurrent(&primary_cuda_context);
int gpu_id;
cudaGetDevice(&gpu_id);
sl::InitParameters init_parameters;
init_parameters.sdk_cuda_ctx = primary_cuda_context;
init_parameters.sdk_gpu_id = gpu_id; ```

Why is setting `sdk_gpu_id` required?

obraun-sl, 2019-06-12:
that looks correct. `sdk_gpu_id` is only required if you have multiple NVIDIA gpu in your PC and want to select one.

To use the primary context, cuCtxGetCurrent() must be used (as you did) to get the current context. If not  cuCtxCreate was called before, the primary context will be used.

fishcu, 2019-06-12:
Thank you for your help!

I only have a single GPU in my system.
If I don't specify `sdk_gpu_id`, I get the error of `NO GPU COMPATIBLE` when trying to open the camera.
The documentation also hints that setting `sdk_gpu_id` is required when specifying a CUDA context manually -- which would explain the above.

I'm just wondering why that is the case.

obraun-sl, 2019-06-12:
Right, that's correct. Since we need to call internally cudaSetDevice to make sure that the current context you provide is correctly attached to the gpu id.
That's more an internal security. On your side, you could set sdk_gpu_id = 0 as well.

fishcu, 2019-06-12:
Okay, thank you.

From a usability perspective, I would probably prefer if the ZED SDK would use the primary context by default, instead of creating and destroying its own -- as that can lead to confusing errors as we have seen. I'm sure you have your reasons for doing it that way though.

Thanks again. Closing.

closed 2019-06-12 by fishcu