stereolabs/zed-sdk #154: Export SVO file to avi UNRECTIFIED

FabianAC07, 2019-06-13:
Hello all, 

Is there a way to export the SVO file video to .avi  WITH OUT rectification?

Thanks!

obraun-sl, 2019-06-14:
Hi,
Yes you can change the VIEW_MODE of retrieveImage (https://github.com/stereolabs/zed-examples/blob/master/svo%20recording/export/src/main.cpp#L159) and select VIEW_LEFT_UNRECTIFIED instead of VIEW_LEFT for example.

FabianAC07, 2019-06-14:
Hi @obraun-sl, 

I have made the changes on the main.cpp file in the lines 159 and 163. 

Now while I am trying to build the file in CMake I get the same error as many other users: 

[image]

I followed these issues where they had the same problem: 
1. https://github.com/stereolabs/zed-examples/issues/21
2. https://github.com/stereolabs/zed-examples/issues/127

As you suggested on issue 127, I manually changed the **OpenCV_CONFIG_PATH** to my installed version of OpenCV 3.4.5,`C:\OpenCV-3.4.5\opencv-3.4.5\build` as follow:

[image]

When I hit Configure bottom again, I get the same error. 

[image]

How to solve this? I am not an expert on programming, so any help would be appreciated!

closed 2020-07-06 by FabianAC07