stereolabs/zed-sdk #157: Win10 failed to setFromStream with the cameras were connected with a TX2 board

wkoa, 2019-06-27:
I am trying to get a stream from two remote cameras connected with one TX2 board. I also download zed-examples and make it successfully. 
When I trying to run receiver in win10(VS 2015) and sender in TX2, the receiver program failed to detect the camera. In the sender terminal, I can see the message: AddingReceiver with IP:192.168.1.102
In the TX2 Board(Ubuntu 16), I run sender and receiver(with ip127.0.0.1 and port 30000) together, and it works.
Both TX2 and Win10 have CUDA9 and SDK2.8.2 with Firmware is 1142.
Any ideas for me?

obraun-sl, 2019-06-27:
Hi,
Make sure that the port is open on windows. Normally you should have a popup window when your sample start to authorize the application to access the network.
Best,
OB

wkoa, 2019-06-27:
> Make sure that the port is open on windows. Normally you should have a popup window when your sample start to authorize the application to access the network.

Hi, I have checked out and make sure the port is open on Windows. Unfortunately, it still not work.
[image]

wkoa, 2019-06-27:
I have modified the zed_examples' sender code like that with the receiver is zed_examples' receiver code:
`///////////////////////////////////////////////////////////////////////////
//
// Copyright (c) 2019, STEREOLABS.
//
// All rights reserved.
//
// THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
// "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
// LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR
// A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT
// OWNER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL,
// SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT
// LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE,
// DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY
// THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
// (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
// OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
//
///////////////////////////////////////////////////////////////////////////

/****************************************************************************************
 ** This sample shows how to record video in Stereolabs SVO format.					   **
 ** SVO video files can be played with the ZED API and used with its different modules  **
 *****************************************************************************************/

// ZED includes
#include <sl/Camera.hpp>

// Sample includes
#include "utils.hpp"

// Using namespace
using namespace sl;

int main(int argc, char **argv) {
    // Create a ZED camera
    Camera zed;
    std::vector<sl::DeviceProperties> devList = sl::Camera::getDeviceList();
    int nb_detected_zed = devList.size();
    std::cout << " Number of ZED Detected : " << nb_detected_zed << std::endl;
    // Set configuration parameters for the ZED
    InitParameters initParameters;
    initParameters.camera_resolution = RESOLUTION_HD1080;
    initParameters.depth_mode = DEPTH_MODE_NONE;
    initParameters.camera_fps = 15;
    initParameters.sdk_verbose = true;

    initParameters.sdk_verbose_log_file = "./zed_log.log";
    // Open the camera

    initParameters.input.setFromCameraID(0);
    ERROR_CODE err = zed.open(initParameters);
    if (err != SUCCESS) {
        std::cout << toString(err) << std::endl;
        zed.close();
        return -1; // Quit if an error occurred
    }
    initParameters.input.setFromCameraID(1);
    Camera zed_1;
    err = zed_1.open(initParameters);
    if (err != SUCCESS) {
        std::cout << toString(err) << std::endl;
        zed_1.close();
        return -1; // Quit if an error occurred
    }

    sl::StreamingParameters stream_params;
    stream_params.codec = sl::STREAMING_CODEC_AVCHD;
    stream_params.port = 25000;
    stream_params.bitrate = 12500;
    stream_params.adaptative_bitrate = true;
    stream_params.gop_size = -1;
    err = zed.enableStreaming(stream_params);
    if (err != SUCCESS) {
        std::cout << "Streaming initialization error. " << toString(err) << std::endl;
        zed.close();
        return -2;
    }
   
    stream_params.port = 30000;
    err = zed_1.enableStreaming(stream_params);
    if (err != SUCCESS) {
        std::cout << "Streaming initialization error. " << toString(err) << std::endl;
        zed.close();
        return -2;
    }

    SetCtrlHandler();

    int fc = 0;
    while (!exit_app) {
        if (zed.grab() == SUCCESS && zed_1.grab() == SUCCESS) {
            sl::sleep_ms(1);
            fc++;
        }
    }

    // Stop recording
    zed.disableStreaming();
    zed_1.disableStreaming();
    zed.close();
    zed_1.close();
    return 0;
}`

obraun-sl, 2019-06-27:
* Can you try with a reduced bitrate (2000 for example) ?
* Can you double check that the receiver side has the correct IP (the one from the jetson) in setFromStream() ?

wkoa, 2019-06-28:
I have used the Router's LANs to connect two devices and it was failed to transport streaming.
However, when I use a Switch to connect two devices, it works.
So Does Switch or Router really influence the camera streaming？ @obraun-sl

wkoa, 2019-06-28:
It seems I should close the windows's defender. Then, it works.
THX.

closed 2019-06-28 by wkoa