stereolabs/zed-sdk #159: Convert svo file to only left camera .avi file

rishabh700, 2019-07-03:
Hi, I know we can convert .svo format to .avi format with visuals from both the lenses. But, I would like to convert the .svo into .avi with visuals from only the left camera. Is this possible?

obraun-sl, 2019-07-04:
Hi,

You can use this samples https://github.com/stereolabs/zed-examples/tree/master/svo%20recording/export
and save only the left image provided by zed.retrieveImage(left, sl::VIEW_LEFT).

rishabh700, 2019-07-05:
Do you mean that I need to make changes in the exprt\src\main file and then build the application ZED_SVO_Recording again using cmake? If yes, should I make a new input parameter (say '5') which would be responsible to provide .avi with only left lens video?

obraun-sl, 2019-07-08:
Yes. You need to change main.cpp to take the left image only and rebuild the sample. If you want you can add a new parameter for that.

hankhhub, 2019-10-14:
Hi, I am also working on the similar problem and I was able to build the application to output only left camera view. I realize the output videos are rotated 90 degrees before and after the new build. Why is that so?

rishabh700, 2019-10-15:
I have no idea why are you getting a rotation. The trick I used to convert SVO into the left only AVI video is to first extract all the images in sequence using the existing SVOExport application. Then use a simple video writer and combine all the image sequence into a video (eg: [here](https://www.life2coding.com/convert-image-frames-video-file-using-opencv-python/)). 
Using this method, you can also get depth only and right only videos in the AVI file format.

casperbh96, 2021-06-04:
For anyone who wants something like this, I created a minimal working example in Python. Works the same as with the svo_export.py script with same arguments

```python
import sys
import pyzed.sl as sl
import numpy as np
import cv2
from pathlib import Path
import traceback

def progress_bar(percent_done, bar_length=50):
    done_length = int(bar_length * percent_done / 100)
    bar = '=' * done_length + '-' * (bar_length - done_length)
    sys.stdout.write('[%s] %f%s\r' % (bar, percent_done, '%'))
    sys.stdout.flush()

try:
    # Get input parameters
    svo_input_path = Path(sys.argv[1])
    output_path = Path(sys.argv[2])
    output_as_video = True    

    # Specify SVO path parameter
    init_params = sl.InitParameters()
    init_params.set_from_svo_file(str(svo_input_path))
    init_params.svo_real_time_mode = False  # Don't convert in realtime
    init_params.coordinate_units = sl.UNIT.MILLIMETER  # Use milliliter units (for depth measurements)

    # Create ZED objects
    zed = sl.Camera()

    # Open the SVO file specified as a parameter
    err = zed.open(init_params)
    if err != sl.ERROR_CODE.SUCCESS:
        sys.stdout.write(repr(err))
        zed.close()
        exit()

    # Get image size
    image_size = zed.get_camera_information().camera_resolution
    width = image_size.width
    height = image_size.height

    # create video object
    video_writer = cv2.VideoWriter(
        str(output_path),
        cv2.VideoWriter_fourcc('M', '4', 'S', '2'),
        max(zed.get_camera_information().camera_fps, 25),
        (width, height)
    )

    # parameters
    rt_param = sl.RuntimeParameters()
    rt_param.sensing_mode = sl.SENSING_MODE.FILL
    nb_frames = zed.get_svo_number_of_frames()

    # Prepare single image containers
    left_image = sl.Mat()
    #right_image = sl.Mat()
    #depth_image = sl.Mat()

    # empty RGBA image
    img_rgba = np.zeros((height, width, 4), dtype=np.uint8)

    while True:
        if zed.grab(rt_param) == sl.ERROR_CODE.SUCCESS:
            svo_position = zed.get_svo_position()

            # Retrieve SVO images
            zed.retrieve_image(left_image, sl.VIEW.LEFT)
            #zed.retrieve_image(right_image, sl.VIEW.RIGHT)
            #zed.retrieve_image(depth_image, sl.VIEW.DEPTH)

            if output_as_video:
                #img_right = right_image.get_data()
                #img_depth = depth_image.get_data()
                img_rgba = left_image.get_data()
                img_out = cv2.cvtColor(img_rgba, cv2.COLOR_RGBA2RGB)
                video_writer.write(img_out)

        progress_bar((svo_position + 1) / nb_frames * 100, 30)

        # Check if we have reached the end of the video
        if svo_position >= (nb_frames - 1):  # End of SVO
            sys.stdout.write("\nSVO end has been reached. Exiting now.\n")
            break

except Exception as ex:
    traceback.print_exc()
finally:
    zed.close()
```

luigipenco, 2021-12-15:
You can just modify the main.cpp file of the export cpp example.
I edited just few lines of code as follows:
`cv::Mat image_single;
    if(output_as_video){
        //image_sideByside = cv::Mat(image_size.height, image_size.width *2, CV_8UC3);
        image_single = cv::Mat(image_size.height, image_size.width *1, CV_8UC3);
    }`
    
    `int frame_rate = fmax(zed.getInitParameters().camera_fps, 25); // Minimum write rate in OpenCV is 25
        //video_writer.open(output_path, fourcc, frame_rate, cv::Size(image_size.width*2, image_size.height));
         video_writer.open(output_path, fourcc, frame_rate, cv::Size(image_size.width*1, image_size.height));`
         
  `cv::cvtColor(left_image_ocv, image_single(cv::Rect(0,0,image_size.width,image_size.height)), cv::COLOR_BGRA2BGR);
                // cv::cvtColor(left_image_ocv, image_sideByside(cv::Rect(0,0,image_size.width,image_size.height)), cv::COLOR_BGRA2BGR);
                // cv::cvtColor(right_image_ocv, image_sideByside(cv::Rect(image_size.width,0,image_size.width,image_size.height)), cv::COLOR_BGRA2BGR);`

closed 2022-04-26 by github-actions[bot]