stereolabs/zed-sdk #160: camera streaming receiver occurred NO GPU COMPATIBLE

sewonist, 2019-07-07:
My test environment are like below that.

[Windows]
CPU : Intel(R) Core(TM) i5-4690 CPU @ 3.50GHz
RAM : 16Gb
GPU : GeForce GTX 970 (4GB)
ZED : 2.8.2
CUDA : 10.0.130

[Jetson Nano]
CPU : ARMv8 Processor rev 1
RAM : 4Gb
GPU : NVIDIA Tegra X1
ZED : 2.8.2
CUDA : 10.0.166

I just set up camera sender at Jetson and set up camera receiver at Windows. The camera receiver occurred NO GPU COMPATIBLE when I try to connect sender in Jetson. However, all tools working are fine with Windows system only. I couldn't imagine what I am wrong?? Please just let me know what can I do for fix.

Thanks a lot

obraun-sl, 2019-07-08:
Hi,
Can you run the receiver with InitParameters::sdk_verbose option at true, so that you can have more info in the console output.
One of the explanation can be that you are using HEVC (H265) codec for streaming and the GT970 does not support H265 decoding (--> NO GPU COMPATIBLE).

sewonist, 2019-07-10:
> Hi,
> Can you run the receiver with InitParameters::sdk_verbose option at true, so that you can have more info in the console output.
> One of the explanation can be that you are using HEVC (H265) codec for streaming and the GT970 does not support H265 decoding (--> NO GPU COMPATIBLE).

Thanks you for reply, I used default sample files. It has written sdk_verbose option at true already. I try to some option that change streaming codec and camera resolution. Unfortunately, it doesn't work until now. 

```ZED (Init) >> Depth mode: NONE 
ZED (Init) >> Video mode: HD720@60
Failed to query video capabilities: Inappropriate ioctl for device
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 4 
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4 
H264: Profile = 77, Level = 50 
 streaming is in progress.... ```

I will try to other GPU though I should use GT970 also. If you have any idea, Please let me know.

Thanks again.

obraun-sl, 2019-07-10:
Hi, 
About the InitParameters::sdk_verbose, I was more talking about the *receiver* (on windows) to see why the GPU is not compatible/detected.
Did you try on localhost as well ? (Windows to Windows, same PC) ?

sewonist, 2019-07-11:
I try from Jetson NANO to Windows. I think I have two issue. One is that I couldn't change port. Other is I can see below error coded.

`Failed to query video capabilities: Inappropriate ioctl for device`

I just guess theses issue make problem.

[image]

obraun-sl, 2019-07-11:
This line 
` Failed to query video capabilities: Inappropriate ioctl for device `
is not important (just a warning from the encoder)

What do you mean by "you cannot change port" ? Inside the sample ?
make sure you have this line : https://github.com/stereolabs/zed-examples/blob/master/camera%20streaming/sender/src/main.cpp#L56

sewonist, 2019-07-11:
I know how to change port in sample. It works fine on Windows. However, it doesn't work on Jetson. You can see below image. I set port to 8052 but it didn't change.

[image]

As you mention the warning is not important, I really don't know why 'camera streaming' sample doesn't work. Sample works fine on Windows, same PC but it doesn't work with Jetson NANO.

sewonist, 2019-07-15:
[image]

Today I tested sender and receiver on one Jetson NANO. I have same problem. 

Jetson [Sender] --- [Receiver] Jetson         <== NO GPU COMPATIBLE
Jetson [Sender] --- [Receiver] Windows     <== NO GPU COMPATIBLE
Windows [Sender] --- [Receiver] Windows <== OK

I doubt ZED SDK for Jetpack 4.2

obraun-sl, 2019-07-18:
If you stream on port 8052 , then the receiver should be on 127.0.0.1:8052. 
But prefer a higher port to avoid conflicts. For example : 
`./ZED_Streaming_sender 40000 && ./ZED_Streaming_Receiver 127.0.0.1:40000`
Note that until a client (receiver) has not requested a connection, the sender will not send packet, therefore it does not appears in the network event list.

phamdat09, 2019-12-23:
Dear,
I also have same problem.

Following your code for "sender", when i run this line: `err = zed.enableStreaming(stream_params);`, it raise one error as : "Cannot load library HEVC". After that, it raise "NO GPU COMPATIBLE
"
I run this code on Jetson nano with newest SDK (2.8.4).
Anyone can fix it ?
Thanks!!!

closed 2022-04-26 by github-actions[bot]

lukelmg, 2025-09-17:
Also having this issue:

```
ironton@ironton-jetson:~/testing-spatial$ python zed_stream.py --resolution HD720
[Sample] Using Camera in resolution HD720
[2025-09-17 22:57:02 UTC][ZED][INFO] Logging level INFO
[2025-09-17 22:57:03 UTC][ZED][INFO] [Init]  Depth mode: NONE
[2025-09-17 22:57:04 UTC][ZED][INFO] [Init]  Camera successfully opened.
[2025-09-17 22:57:04 UTC][ZED][INFO] [Init]  Camera FW version: 1523
[2025-09-17 22:57:04 UTC][ZED][INFO] [Init]  Video mode: HD720@30
[2025-09-17 22:57:04 UTC][ZED][INFO] [Init]  Serial Number: S/N 33503129
Streaming on port  30000
[Streaming] Use Transport layer mode : 0
[ZED][Encoding] Critical Error : No Video Enc [enc_0]
[ZED][Streaming] Failed to create encoder [0.0] at resolution 1280x720
[2025-09-17 22:57:05 UTC][ZED][ERROR] [Streaming]  Cannot enable streaming: Err Code -4
Streaming initialization error:  NO GPU COMPATIBLE
```