stereolabs/zed-sdk #161: The image is inverted and the color is not normal.

guan4286, 2019-07-13:
My operate system: Ubuntun 16.04 cuda9.0 zed_sdk2.8.
when I open ZED Explorer ,I recieved the following image ：

guan4286, 2019-07-13:
[image]

guan4286, 2019-07-13:
[image]

guan4286, 2019-07-13:
how should to solve this problem ？

guan4286, 2019-07-13:
[image]
this is my cuda and nvdia

obraun-sl, 2019-07-15:
This comes from a non compatible OpenGL driver version (GLSL error).
Make sure that opengl driver (mesa) are installed or during NVIDIA driver installation, make sure that it installs the opengl library.

closed 2022-04-26 by github-actions[bot]