stereolabs/zed-sdk #162: streaming video is not working.

djkayip, 2019-07-18:
Hi, I'm trying to stream the video. but it seems like streaming video is not stable. 
At receiver, even though I run a same code. it returns very different errors.

1.(ZED Stream) Received unexpected size pack 144/65016 

or 

2(ZED Stream) FAILED: Stream opening TIMEOUT

I try to reduce bitrate, and increase gop_size. there has been no difference.

I'm using Python API, CUDA 10.

obraun-sl, 2019-07-18:
This is probably coming from lost packet in the streaming (lost packet means corrupted images until a new IDR frame is reached, depending on the gop size).
Did you try to stream on localhost to check if you have the same behavior ?

djkayip, 2019-07-19:
streaming on localhost(127.0.0.1) works ok. but if I try to stream between two PCs with a same router, it produces the problem. I tried to increase gop_size up to 256, and change the bitrate (from 1000-10000).  
it doesn't seem to make any difference. 

it there anything I can do?

obraun-sl, 2019-07-19:
Do you have something else connected on the same router? or only the two PCs?
* Increase gop size and decrease bitrate will reduce the number of lost/corrupted packet.
* Make sure all the ethernet cables are at least cat5E.

djkayip, 2019-07-19:
Yes, Only two PCs are on the same router. I did try to increase gop size and decrease bitrate. it still produces same problem.  Do you mean I need to change the ethernet cable to the wifi router?

khausman502, 2019-11-14:
Did you ever come to a resolution for this issue?  I'm facing the exact same problem--streaming works fine on a single machine, but between two machines, I get the macro-blocking during any motion.  Have verified that bandwidth between the machines is sufficient.  I also see that setting the bitrate on the sending side seems to have no effect on the bandwidth I see flowing over the wire.

templarares, 2020-01-13:
Is there any update/fix regarding this issue yet? I have run into the same issue. Streaming (720Px30FPS) works fine with an ethernet cable, but it became extemely laggy when the cable is replaced with wireless connection, whose bandwidth has been verifited to be sufficient (25Mbps). I, too, observed that varying the birtate seems to have no effect on the ethernet bandwith consumption monitered by Win10's task manager.

obraun-sl, 2020-01-14:
Hi,
streaming module has been reworked for next major release 3.0.0. This release will be official in the next weeks (end of January). 
That brings more stability and a payload size parameter to optimize the bandwith consumption. 

Wifi is known to brings more collision into stream packet than ethernet cable. Try also to reduce the number of PC connected to the same Wifi Network.

templarares, 2020-01-15:
Hi Obraun-sl,

Thank you very much for your quick response! And there are some more specific symptoms observed in our tests; as we do not know about the lower-level structure of the "camera streaming" sender sample, do you think they are expected given the current coding in "camera streaming" sender and that the 3.0.0 release would fix them? Thank you very much!

#1. With DEPTH_MODE_NONE, 720P x 30FPS (x 2 Cameras) configured in "camera streaming" sender , the bandwidth consumption is 17Mbps, which is unexpectedly high: normally, with H.265, 720Px30FPS should consume no more than 2 Mbps;

#2.Changing the "bitrate" parameter in "camera streaming" sender  does not seem to take any effect; the bandwidth consumption with the configuration mentioned in #1 stays at 17Mbps regardless to the bitrate value (be it 8000kbps, 4000kbps or 1000kbps).

#3.Our wireless transmitter has working bandwith of 25Mbps (at this bandwith, the ping latency varys from 60ms to 120ms, averaged at about 75ms), which is greater than the 17Mbps mentioned in configuration #1. However, the ZED Unity Plugin cannot display the video stream (screen stays at "Loading camera"). By reducing the FPS from 30 to 15, the "camera straming"'s bandwith consumption is reduced to 9Mbps as expected; at this configuration, the ZED Unity Plugin finally displays the images, but extremely laggy. Do you think this is also caused by the coding in current "camera streaming" sender and will be fixed in the 3.0.0 release? And is there any debug log we can check at the ZED Unity Plugin side to see if we can address the issue from the Receiver's end?

Additionally, we are also considering developing our own solution to make the "TypeC-USB3.0" cable between ZED_M and the PC go wireless. May I know the format of the data stream coming out of the TypeC port/out of the USB3.0 port? And what about the data stream's approximate bitrate, say, with DEPTH_MODE_PERFORMANCE, 720P, 30FPS? 

Thank you very very much!

muraiyuya, 2020-06-22:
Have you already solved this issue? I'm facing the same problem, between two machines, I get the macro-blocking when the camera moves. I want streaming-video to be much smoother .

templarares, 2020-06-24:
> Have you already solved this issue? I'm facing the same problem, between two machines, I get the macro-blocking when the camera moves. I want streaming-video to be much smoother .

After updating the ZED SDK to 3.0+, video streaming now works, wirelessly. But my ZED is stationary (objects in the scene do move, though).

Zgihu, 2020-06-24:
I see.That is good.
In my case, video streaming is so smooth when pc with zed is connected by LAN cable.
However, video streaming stops sometimes when pc with zed is connected by wireless LAN .
How can I solve it? 
If you know it, please teach me!

closed 2022-04-26 by github-actions[bot]

Rutherfordum, 2022-05-15:
I'm facing the same problem, between two machines, I get the macro-blocking when the camera moves. I want streaming-video to be much smoother . Help.

obraun-sl, 2022-05-16:
This can come from two sides : 
- Loss of stream packet. If you are in verbose mode, on the receiver side, you will see some message in console (unexpected size.....) when packets are lost. If this is the case, this most probably comes from the network. Try to use ethernet instead of WiFi, and try to play with the streaming parameters "chunk_size" (increase it most probably to reduce the number of packet for an image). 

- Try to increase the bitrate if there is now corrupted packet but just H26x compression artifacts when moving.