stereolabs/zed-sdk #164: On Ubuntu 18 with CUDA 10.0 fail: "ERROR: Found Cuda 9.1 but the ZED SDK requires 10.0"

EmilPi, 2019-07-24:
CUDA is installed in /usr/local/cuda10-0.
Computer hardware is AMD Ryzen 2700X + NVidia RTX 2080.
On this PC, neural networks libraries successfully work with CUDA 10.0.

adujardin, 2019-07-24:
Unfortunately, there's currently no compatible version of the ZED SDK with CUDA 10.1, only CUDA 10.0.
They are not binary compatible, the required version must exactly match.

EmilPi, 2019-07-31:
For those who will experience this kind of problems, on fresh ubuntu system make sure to follow this guide until `sudo apt-get install cuda` command: https://medium.com/@zhanwenchen/install-cuda-10-1-and-cudnn-7-5-0-for-pytorch-on-ubuntu-18-04-lts-9b6124c44cc

closed 2019-07-31 by EmilPi