stereolabs/zed-sdk #165: ZED camera causes ubuntu 18.04 with RX 2080 and CUDA 10.0 to freeze

EmilPi, 2019-07-31:
**********
My setup:

Ubuntu 18.04
AMD Ryzen 2700X
NVidia RTX 2080
nvidia-driver-410
cuda-10-0

**********
Steps to reproduce:
1)
Connect ZED camera
If ZED camera is not detected, try plug/unplug until it is
Try to calibrate camera
if the calibartion window will become unresponding, try restarting calibration
Wait for system freeze

2)
Connect ZED camera
If ZED camera is not detected, try plug/unplug until it is
Try to see video using ZED explorer
if camera is not detected, just wait

obraun-sl, 2019-08-08:
Hi,
Can you tell the ZED SDK version you are using ?

EmilPi, 2019-08-08:
The latest, 2.8.3, from https://download.stereolabs.com/zedsdk/2.8/ubuntu18 .

closed 2022-04-26 by github-actions[bot]