stereolabs/zed-sdk #171: how to convert cv::Mat to sl::Mat

wangzhongju, 2019-08-26:


obraun-sl, 2019-08-27:
Hi,

You can take a look here : 
https://github.com/stereolabs/zed-examples/blob/master/svo%20recording/export/include/utils.hpp#L64

For OpenCV Mat to SL Mat, you can use the same idea.

mrlalala, 2022-04-25:
> Hi,
> 
> You can take a look here : https://github.com/stereolabs/zed-examples/blob/master/svo%20recording/export/include/utils.hpp#L64
> 
> For OpenCV Mat to SL Mat, you can use the same idea.

Hello, this link is no longer available now. Could you please update it again? Thanks a lot!

obraun-sl, 2022-04-25:
Here it is : 
https://github.com/stereolabs/zed-examples/blob/master/svo%20recording/export/cpp/include/utils.hpp#L66

Code as reference: 

```
cv::Mat slMat2cvMat(sl::Mat &input) {
    int cv_type = -1;
    switch (input.getDataType()) {
        case sl::MAT_TYPE::F32_C1: cv_type = CV_32FC1; break;
        case sl::MAT_TYPE::F32_C2: cv_type = CV_32FC2; break;
        case sl::MAT_TYPE::F32_C3: cv_type = CV_32FC3; break;
        case sl::MAT_TYPE::F32_C4: cv_type = CV_32FC4; break;
        case sl::MAT_TYPE::U8_C1: cv_type = CV_8UC1; break;
        case sl::MAT_TYPE::U8_C2: cv_type = CV_8UC2; break;
        case sl::MAT_TYPE::U8_C3: cv_type = CV_8UC3; break;
        case sl::MAT_TYPE::U8_C4: cv_type = CV_8UC4; break;
        default: break;
    }
    // Since cv::Mat data requires a uchar* pointer, we get the uchar1 pointer from sl::Mat (getPtr<T>())
    // cv::Mat and sl::Mat will share a single memory structure
    return cv::Mat(input.getHeight(), input.getWidth(), cv_type, input.getPtr<sl::uchar1>(sl::MEM::CPU));
}
```

closed 2022-06-02 by github-actions[bot]