stereolabs/zed-sdk #172: Issues with depth data

sriyan983, 2019-08-29:
Attached are screenshots of the output.

I am using pytorch and darknet with yolov3-SPP. You can check this repo to understand the implementation.
https://github.com/ultralytics/yolov3.

Following is my code to detect an object from a ZED feed and use the MEASURE XYZ to get the distance to object.
https://pastebin.com/UwRAyj8X

[Depth calculation logic can be seen: getObjectDepth()]

The person standing further seems to be correct but the one sitting and working (me), the distance results are too inaccurate (+2m).

Can you please check the code below and see if I am doing something wrong or this is how it is supposed to be?

[image]
[image]

adujardin, 2019-08-30:
You're currently using the fastest, and therefore less robust depth mode. You can replace `sl.DEPTH_MODE.DEPTH_MODE_PERFORMANCE` l.248 by `sl.DEPTH_MODE.DEPTH_MODE_ULTRA`. It's heavier to compute but the accuracy is increased.

To get an idea of the depth quality, you can use the DepthViewer tool which includes a 3D view of the point cloud with different settings to see the effect.

The code for depth extraction seems correct. However, it is focused on the center, you could try to sample a larger area (and skip some pixels, like 1 every 4 pixels) to get a more robust value.
Some more advanced method would be to compute the depth histogram on the ROI defined by the box.

sriyan983, 2019-08-31:
Thanks a bunch for your response. I shall try the same and post the results.

rishabh700, 2019-09-02:
You can also try to filter the depth values inside the sample area based on the confidence threshold values and then take the median of all the x,y,z values.

closed 2022-04-27 by github-actions[bot]