stereolabs/zed-sdk #178: SVO video to Audio synchronization issue

sneheshs, 2019-09-12:
Hi, I am using a clapper and a separate recording microphone to record audio. When I try to synchronize the audio, the beginning time for video and audio works out well and simply trim the video. However, as the video goes on, it seems the lag between audio and video gets larger and larger. In a 5 minute video, the lag was as long as 5 seconds by the end.

BTW I am recording multiple ZED via USB 3.1 and saving them to fast USB 3.0 SSD with LOSSLESS mode. My GPU only has 2 encoders so I am not able to use compression.

1. Have you guys ever tested or could you test on your end if the fps is in fact consistent? 
2. How can I find out frames dropped?
3. I simply modified your SVO multi-camera recording example to record 3 ZEDs at 720p 30fps.
4. And I modified your zed export example, to export SVO to MP4.

Please let me know if you can think of any other reasons this might be happening that I should check.

obraun-sl, 2019-09-13:
Hi,
LOSSLESS mode is the most expensive mode in terms of performance, since it requires compression time + writing time. 
You can try LOSSY mode and/or AVC/HEVC compression mode which are more effective. You will have less to no frame drops.

You can also handle the frame drops by comparing frame timestamps between frame n and frame n+1 and delay the reading loop.

sneheshs, 2019-09-13:
Hi Oliver, thanks for your response. The LOSSLESS doesn't do any compression based on the documentation, so it is straight writing to disk. Since I am trying to use 3 cameras and my GPU only support up to 2 encoders, I was not able do that. Maybe I could try using AVC/HEVC for two cameras and LOSSLESS for one. I have not tried that yet. But I was just surprised that even for a single camera at LOSSLESS it was dropping packets.

I will try out the timestamps between frame n and n+1 using the frame timestamps. Thanks.

closed 2022-04-27 by github-actions[bot]