stereolabs/zed-sdk #179: ZED (Mesh) >> Error : Not enough memory to apply the textures on the Mesh

FlorianHergo, 2019-09-16:
Hello,

On a Nano jetson we need to specify the maximum size (max_memory_usage = 1100) of memory to allocate one to spacial mapping.

But we also have a potential 6gig SWAP memory by doing this: https://github.com/JetsonHacksNano/installSwapfile

However you do not take it into account. And during the recording or even during the scan, the memory saturates and we have this error: 

ZED (Mesh) >> Error : Not enough memory to apply the textures on the Mesh.

Can you tell me if there is a trick or if I do wrong.

Thx

FlorianHergo, 2019-09-20:
Hi @obraun-sl ,

Have you read this message, this is the most important bug for us on Nano Jetson. Memory management must either be optimized directly in the SDK, or give us the hand to manage the direction of the memory call?

What do you think ?

obraun-sl, 2019-09-23:
Hi,
The spatial mapping requires quite a lot of memory to be able to store all the data to generate the mesh and the nano does not have a lot. 
-> Have you tried to divide the mapping into small mapping chunks to avoid memory increase (save the obj file once memory is increasing too much) ? Then Merge all the mesh into a single one with mesh fusion tools.

To see how we can improve the memory usage, are you using the spatial mapping sample or a modified one? (just to make sure we are on the same basis).

FlorianHergo, 2019-09-23:
Hello,
Yes exactly on the example you are talking about GIT. No modification.

Saving .Obj at the same time is not the problem, even if it can take a very long time (15 to 20min) it never crashes. I'm only talking about the part : to apply the textures on the Mesh.

Except error on my part, but when I look at your code master, I notice that you do not direct the possibility of the use of SWAP. Only the use of physical RAM. It is therefore for me the problem.

And apparently you have already had requests of this type from other users, but the messages have no answer. I have the impression that this memory write concern is more global in your SDK than only for MESH textures.

thanks.

closed 2022-04-27 by github-actions[bot]