stereolabs/zed-sdk #180: Jetson TX2, Zed Mini, nvbuf_utiks: Could not get EGL display connection , Segmentation fault

mesimolnar, 2019-09-17:
Hi! 

I would like to use Zed Mini with Jetson TX2 to stream localhost to use an outer IP streaming. But, if I start the receiver I got the message: 
`nvbuf_utils: Could not get EGL display connection
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
ZED (Init) >> Depth mode: PERFORMANCE
libv4l2_nvvidconv (0):(802) (INFO) : Allocating (12) OUTPUT PLANE BUFFERS Layout=1
libv4l2_nvvidconv (0):(818) (INFO) : Allocating (12) CAPTURE PLANE BUFFERS Layout=0
Error: (30): unknown error
ERROR: while compiling Shader :
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

ERROR: while compiling vertex shader
ERROR: while compiling Shader :
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

ERROR: while compiling fragment shader
Segmentation fault (core dumped)
`
Have you any idea to solve this problem?

mesimolnar, 2019-09-18:
Hi, I solved the problem vith the version, and after that I got another problem with the precision, and I solved it, BUT, az got de problem also 

> Error: (30): unknown error
Segmentation fault (core dumped)

Have you any idea what is the problem with that, and how can I solve it?

obraun-sl, 2019-09-18:
Can you post the code you are using that generated this error ?
What is the Jetpack version you are using ?

mesimolnar, 2019-09-18:
Hello.
I use Jetpack 4.2.1(rev2) .
I have only modified the GLViewer.cpp. This is the modified code: 

> #include "GLViewer.hpp"

GLchar* VERTEX_SHADER =
"#version 300 es\n"
"precision mediump float;\n"
"layout(location = 0) in vec3 in_Vertex;\n"
"layout(location = 1) in vec3 in_Color;\n"
"uniform mat4 u_mvpMatrix;\n"
"out vec3 b_color;\n"
"void main() {\n"
"   b_color = in_Color;\n"
"	gl_Position = u_mvpMatrix * vec4(in_Vertex, 1);\n"
"}";

GLchar* FRAGMENT_SHADER =
"#version 300 es\n"
"precision mediump float;\n"
"in vec3 b_color;\n"
"layout(location = 0) out vec4 out_Color;\n"
"void main() {\n"
"   out_Color = vec4(b_color, 1);\n"
"}";

GLViewer* currentInstance_ = nullptr;

GLViewer::GLViewer() : available(false){
    currentInstance_ = this;    
    mouseButton_[0] = mouseButton_[1] = mouseButton_[2] = false;
    clearInputs();
    previousMouseMotion_[0] = previousMouseMotion_[1] = 0;
}

GLViewer::~GLViewer() {}

void GLViewer::exit() {
    if (currentInstance_) {
        pointCloud_.close();
        available = false;
    }
}

bool GLViewer::isAvailable() {
    if(available)
        glutMainLoopEvent();
    return available;
}

Simple3DObject createFrustum(sl::CameraParameters param) {

    // Create 3D axis
    Simple3DObject it(sl::Translation(0, 0, 0), true);

    float Z_ = -150;
    sl::float3 cam_0(0, 0, 0);
    sl::float3 cam_1, cam_2, cam_3, cam_4;

    float fx_ = 1.f / param.fx;
    float fy_ = 1.f / param.fy;

    cam_1.z = Z_;
    cam_1.x = (0 - param.cx) * Z_ *fx_;
    cam_1.y = (0 - param.cy) * Z_ *fy_;

    cam_2.z = Z_;
    cam_2.x = (param.image_size.width - param.cx) * Z_ *fx_;
    cam_2.y = (0 - param.cy) * Z_ *fy_;
[... 610 lines cut ...]
    return vertical_;
}

float CameraGL::getZNear() const {
    return znear_;
}

float CameraGL::getZFar() const {
    return zfar_;
}

void CameraGL::updateVectors() {
    forward_ = ORIGINAL_FORWARD * rotation_;
    up_ = ORIGINAL_UP * rotation_;
    right_ = sl::Translation(ORIGINAL_RIGHT * -1.f) * rotation_;
}

void CameraGL::updateView() {
    sl::Transform transformation(rotation_, (offset_ * rotation_) + position_);
    view_ = sl::Transform::inverse(transformation);
}

void CameraGL::updateVPMatrix() {
    vpMatrix_ = projection_ * view_;
}

mesimolnar, 2019-09-18:
And I got the message in terminal when I ran the program : ./Zed_Streaming_Receiver 127.0.0.1

> nvbuf_utils: Could not get EGL display connection
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
ZED (Init) >> Depth mode: PERFORMANCE
libv4l2_nvvidconv (0):(802) (INFO) : Allocating (12) OUTPUT PLANE BUFFERS Layout=1
libv4l2_nvvidconv (0):(818) (INFO) : Allocating (12) CAPTURE PLANE BUFFERS Layout=0
Error: (30): unknown error
Segmentation fault (core dumped)

obraun-sl, 2019-09-20:
Do you have a display connected to your jetson for the receiver part ? According to the console output, it seems not. ` Could not get EGL display connection`

Note that opengl cannot be used if no display is available (impossible to create a GL context then.). The `Error: (30): unknown error` is probably coming from the cuda-opengl interoperability used in the sample 
`checkError(cudaGraphicsGLRegisterBuffer(&bufferCudaID_,bufferGLID_,cudaGraphicsRegisterFlagsNone));`

mesimolnar, 2019-09-20:
No there was no display. Now there is, and it works, thanks.

closed 2019-09-20 by mesimolnar