stereolabs/zed-sdk #182: calibration file missing parameters?

amaanda, 2019-09-19:
When I download the .conf file from http://calib.stereolabs.com/ I got p1 and p2 parameters. But then when I calibrated the camera myself, I don't get these. Why?

This is from the download file provided by StereoLabs:

```
[LEFT_CAM_2K]
fx=1398.04
fy=1398.04
cx=1034.52
cy=668.349
k1=-0.175832
k2=0.028498
p1=0.00105815
p2=0.000436431

[RIGHT_CAM_2K]
fx=1398.26
fy=1398.26
cx=1085.59
cy=648.693
k1=-0.1747
k2=0.0282032
p1=0.000872006
p2=0.000377105

(...)
```

This is what I got from ZED Calibration 2.8.3:
```
[LEFT_CAM_2K]
fx=1399.05
fy=1399.05
cx=1050.25
cy=677.159
k1=-0.166025
k2=0.0250242

[RIGHT_CAM_2K]
fx=1397.4
fy=1397.4
cx=1086.26
cy=658.046
k1=-0.163583
k2=0.0239731

(...)
```

obraun-sl, 2019-09-20:
The factory calibration provides more accurate values for p1/p2 even if very close to 0. The Calibration tool provided in the SDK cannot estimate properly p1/p2 therefore does not take them into account in the model. (p1=p2=0)
But the calibration file can read p1/p2 in any case, therefore if you provide your own calibration ddata with p1/p2, then it will be ingested for sure in the pipeline.

amaanda, 2019-09-20:
How can I calibrate with the same quality from the original calibration, then?

obraun-sl, 2019-09-23:
If you want to have access to all parameters, then you can use opencv calib3D module to perform a stereo calibration. It will compatible results with configuration files (fx,fy,cx,cy, k1, k2, p1, p2, rx, ry==cv , rz, tx==baseline, ty, tz)

closed 2022-04-27 by github-actions[bot]