stereolabs/zed-sdk #187: depth algorithm

soldierofhell, 2019-10-20:
Hi, what kind of algorithm stands behind disparsity estimation? Is it DNN- based? Are there any benchmark with eg. classic OpenCV or KITTI leaderboard? Regards,

snakehaihai, 2019-10-30:
I`m not working with ZED company, but I study stereo as phd research for a while. But based on my understanding and observation. They are using SGM based method with very large P1 and P2. As for cost/Cost aggregation, Cant tell that it is 3 Way sgm or census transform variant. But I dont think it is things like CNN or Daisy Descriptor as those method are way too slow. Hope this gives you some cue.

soldierofhell, 2019-10-30:
Thank you for the info. By NVIDIA you mean their VisionWorks: https://developer.nvidia.com/embedded/visionworks ? Is the code available somewhere (I couldn't find it)?

By the way, what is the reference implementation of SGM? I found a few on github

snakehaihai, 2019-11-22:
yes visionworks. previously it is nvxio. now might be a bit different. i think you can find the detail reference in their webside

soldierofhell, 2019-11-22:
Thank you. I found nvx_demo_stereo_matching example in VisionWorks

closed 2022-04-27 by github-actions[bot]