stereolabs/zed-sdk #188: Cannot open SVO file

hankh1105, 2019-10-22:
Hi, I was trying out the SVO  export utility  with the command **ZED_SVO_Export "path/to/file.svo" "path/to/output/file.avi" 0**, and what I get everytime was 

> ZED (Init) >> ERROR: Cannot load SVO file. Check that the path of the SVO is correct, avoid white spaces in filename, add ".svo" extension to filename.
> INVALID SVO FILE

I made sure the SVO file does exist at the correct path. I tried opening the file using  ZED Explorer. Turns out, it also gave me   **** [SVO] Hardware compression (NVENC) available **
FAILURE : Didn't manage to open svo file INVALID**.

I installed CUDA  10.1 and ZED SDKv2.8.0.

obraun-sl, 2019-10-23:
It may be possible that the SVO file got corrupted. Try to use the -repair option of ZED SVOEditor.

hankh1105, 2019-10-24:
I tried SVOEditor.exe -repair input.svo and I get

>  'SVOEditor.exe' is not recognized as an internal or external command,
> operable program or batch file.

Then I tried **"ZED  SVOEditor".exe -repair input.svo**, it didn't give errors but didn't generate a repaired file either.

obraun-sl, 2019-10-25:
It may be a permission issue then. Try to run your program with admin rights

impermanD, 2022-01-24:
May your file path include some utf-8 code

closed 2022-04-27 by github-actions[bot]