stereolabs/zed-sdk #193: Concatenating multiple sl::mat images

kushal2507, 2019-11-01:
Is there a method to concatenate 2 or more sl::mat images? Something similar to cv::hconcat or cv::vconcat??I dont want to convert my images to cv::Mat images.

I am recording from 2 ZED cameras and want to be able to view the images in a single window.

Myzhar, 2019-11-01:
Hi @kushal2507 
You can directly retrieve SIDE-BY-SIDE images using `VIEW_SIDE_BY_SIDE` as parameter in `retrieveImage`

kushal2507, 2019-11-01:
@Myzhar Thank You for the quick response. I want to concatenate only the left image of both the cameras. From what I understood from the documentation, using VIEW_SIDE_BY_SIDE shows the left and right image of the same camera. 
I am recording each camera using multi-threads, similar to what has been done in the zed-multi-camera repository

Myzhar, 2019-11-01:
Ah, sorry for the misunderstanding.
The only way to achieve your purpose is to create a new image with same height and double width and copy the data line by line using `memcpy` or `std::copy`

closed 2022-04-27 by github-actions[bot]