stereolabs/zed-sdk #194: Get [nan nan nan] while get the point cloud data by using Python API

XiangLi6632, 2019-11-04:
I want to get the 3D coordinate from the point cloud data, but there are many pixels with the value [nan nan nan], how can I solve it? thanks.

obraun-sl, 2019-11-12:
Hi,
This is expected to have Nan values since some pixels may be occluded or just not in the depth range. If you use the STANDARD sensing mode, you need to handle the Nan values and choose correct neighbor values.
Otherwise, another solution is to use the FILL sensing mode that will provide a depth map without holes.

XiangLi6632, 2019-11-12:
> Hi,
> This is expected to have Nan values since some pixels may be occluded or just not in the depth range. If you use the STANDARD sensing mode, you need to handle the Nan values and choose correct [...]

Hi, I tried to set the depth_minimum_distance with 0.3 as the document says, it works, however, it always gives me a warning that the distance is too close, should I ignore it? 
For another question, I want to choose two base point to calculate other objects,  I calculate the distance of these two points in camera coordinate system, it diffs with the distance which I measure in the real world with an error about 5 - 10 mm, and when I choose FILL, the error sometimes even larger, how can I reduce the error, thanks!

obraun-sl, 2019-11-12:
You can ignore the warning if it works. 
FILL always give less precise results because it contains more filtering to fill holes. This results in a less accurate depth on some part of the image (but sometimes easier to use). 
If possible, I would recommend to use the STANDARD sensing mode and the ULTRA depth mode for measurement applications. That's the best combination.

closed 2022-04-27 by github-actions[bot]