stereolabs/zed-sdk #195: Stream and data

mesimolnar, 2019-11-07:
Hi, I would like to use the Zed M camera to stream the video from the camera and the datas also.
I can get the two datatypes from the camera with streaming, but have you any idea if I would like to use the video datas in a fifo. Which datatype I have to convert the video image  to send through a fifo.
My present solution converting the sl::Mat type to cv::Mat and after it encoding to jpg and send fifo, but with this I can't get the good image from the fifo.

Thanks

obraun-sl, 2019-11-12:
Hi,
The jpg encoding might be too heavy for streaming out. 
Instead I would recommend to use gstreamer that would allow you to compress in H26x and send it through RTP or else.

mesimolnar, 2019-11-12:
Hi, ok, I understand, and for the stream I used to use gstreamer, but when I'm streaming with gstreamer, I cannot use the Zed libs to get the datas from the camera. Have you any idea to solve this?

obraun-sl, 2019-11-12:
Can you post the code you used with zed and gstreamer ?

mesimolnar, 2019-11-12:
So my problem is. I can get the video from the camera with gstreamer, and it works fine. But if I try to get the translation and orientation of the camera with the zed libs, the lib says "Camera already in use".
There is any possible way that I can use parallel a gstreamer pipe and the zed lib with the same camera?

obraun-sl, 2019-11-12:
You need to use gstreamer API instead of using the gst-launch program. You cannot access the camera from 2 separated process. Therefore you need to capture the frames using the ZED API, then use gstreamer API to generate the stream.
I think this link can help you : 
https://stackoverflow.com/questions/37339184/how-to-write-opencv-mat-to-gstreamer-pipeline

closed 2022-04-27 by github-actions[bot]