stereolabs/zed-sdk #198: Field of View ZED mini

bizonpatrycja, 2019-11-15:
Hi,

I would like to make sure what the Field of View of ZED mini camera is. First of all, in the description on the Stereolabs website it's written that the "Field of View is: 90° (H) x 60° (V) x 110° (D) max". How should I understand that (I mean the two values of 90 and 110)? I am looking for the FoV of a single camera.

Second of all, on the Stereolabs support page (https://support.stereolabs.com/hc/en-us/articles/360007395634-What-is-the-camera-focal-length-and-field-of-view-) they say with HD2K resolution the horizontal FoV is 76 degrees.

Could someone explain that to me?

Thanks,
Patrycja

obraun-sl, 2019-11-18:
Hi,

90° (H) x 60° (V) x 110° (D) is the filed of view of the full sensor. 
Since the sensor is a 16/9 sensor, the field of view vertical and horizontal are not the same : 
(H) refers to the horizontal field of view.
(V) refers to the vertical
(D) refers to the diagonal fov (commonly used).

Then, depending on the resolution, all the pixels of the sensor might or might not be taken. For example, HD720 and VGA resolution takes the full sensor (or almost) and the resolution is created by binning pixels (4 pixels creates a single one). With this method, the FoV is preserved. 
For some resolution, it's not possible to use this method, and we need to crop the sensor at the desired resolution. This is done for HD2K and HD1080 resolution. A crop will obviously lower the FoV. That is why the FoV in HD2K is a little bit lower that the sensor FoV. (76° vs 90°)

The SDK contains the getCameraInformation() function to have access to the FoV for each resolution (before/after digital alignment).

bizonpatrycja, 2019-11-18:
Hi Oliver,

Thank you for the explanation, it was helpful!

Tetsujinfr, 2020-07-27:
maybe you can consider closing the issue no?

closed 2020-09-01 by bizonpatrycja

luan1412167, 2021-07-29:
Hi @obraun-sl @bizonpatrycja,
Can we change FoV of Zed by python-api?

obraun-sl, 2021-07-29:
See answer here : 
https://github.com/stereolabs/zed-ros2-wrapper/issues/47

(Please don't reopen a closed issue. It's better to open a new one)