stereolabs/zed-sdk #200: support for CUDA 10.1/2?

jwdinius, 2019-12-27:
Any updates on when CUDA>10.0 will be supported by the SDK?

[image]

figo-figo, 2019-12-28:
only support for CUDA 10.0

adujardin, 2020-01-13:
The [2.8.5 is released](https://www.stereolabs.com/developers/release/) and now also support CUDA 10.2

closed 2020-01-13 by adujardin

madoxnet, 2020-05-26:
The latest SD images for Jetson Nano use Cuda 10.2, the latest ZED SDK on https://www.stereolabs.com/developers/release/ doesn't seem to support it.  Its quite annoying to manually install 10.0 and finding an old SD image on Nvidia's site is extremely difficult.

obraun-sl, 2020-05-29:
The support for Jetpack 4.4 will be release with the ZED SDK 3.2 in a few days .
Jetpack 4.3 ( with cuda 10.0) is still the official stable release and you can download it directly here : 
https://developer.nvidia.com/jetpack-43-archive

spiderkeys, 2020-06-04:
Just checking in, since it has been a week since the last comment. Is there an updated timeframe for when SDK 3.2 will be available? Thanks.