stereolabs/zed-sdk #203: Issue on WMR Headset compatibility

jkim010593, 2020-01-14:
Our VRAR lab has multiple WMR headsets. We are currently using Unity 2019.2.17f1. The ZEDMixedRealityPlugin.cs script on line 569, which calls… 
dllz_latency_corrector_get_transform(cameraTimeStamp, useLatency, out t, out r); 
that accesses your v2.8.2 plugin, returns ¬false with the HP Reverb WMR headset but return true with the Samsung Odyssey Plus. 
The most significant difference between these WMR headsets is the Reverb has 4k resolution and the Odyssey has 1080p. 
Can you offer any suggestions on how we might resolve this issue.
Side Note: We researched and found this ticket, and was wondering if there was a solution other than switching headsets.
Link: https://github.com/stereolabs/zed-unity/issues/78

obraun-sl, 2020-01-16:
Hi,
This happens most probably because the Samsung Odyssey does not contains the "windows" or "wmr" in its name which make the latency correction initialize function not called.
You can try by changing the line 
https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/MR/ZEDMixedRealityPlugin.cs#L312
with just a `else` instead of the `else if` to make sure it goes through the initialization.

Note : Since this issue is related to Unity plugin, make sure to post your issue in the dedicated repository , i.e. https://github.com/stereolabs/zed-unity. 

Thanks

jkim010593, 2020-01-20:
That seemed to be the fix for the issue, Thank you.
Another question, is there any chance that your company will be adding this fix in future patches?
So that there is no conflicts for future versions of the SDK?

obraun-sl, 2020-02-03:
Fixed in last master commit 
https://github.com/stereolabs/zed-unity/commit/e80888f6b86c9d2c7b046f1588ec4cfca823043f

closed 2020-02-03 by obraun-sl