stereolabs/zed-sdk #207: Rtabmap make fail - Zed SDK JetPack 4.2 3.0.0

bj-neilson, 2020-01-30:
Hi,
I installed the latest SDK on my TX2 running Jetpack 4.2.  Trying to make RtabMap fails.  Shows' lots of errors relating to StereoZed.  Could this be an SDK problem?

File attached of screen.

Thanks
Ben

[rtabmap_build_fail.txt](https://github.com/stereolabs/zed-examples/files/4132148/rtabmap_build_fail.txt)

adujardin, 2020-01-30:
Hi,
The build failed because Rtabmap isn't compatible with the ZED SDK 3.0 yet, you should downgrade to the SDK 2.8 for now if possible. Related issue https://github.com/introlab/rtabmap/issues/499

bj-neilson, 2020-01-30:
Hi @adujardin ,
Thanks. I only upgraded my SDK installation from 2.8 because when a cloned and tried to build the Ros wrapper I received an error that it required SDK 3.xx or greater. So I guess the wrapper changed too?

adujardin, 2020-01-30:
Yes most official wrapper/samples have been updated to 3.0 but there's a tag for the 2.8 compatible version, for the ROS wrapper you'll find it here https://github.com/stereolabs/zed-ros-wrapper/releases/tag/v2.x

bj-neilson, 2020-01-30:
Thanks. That’ll help!

closed 2020-01-30 by bj-neilson

matlabbe, 2020-01-31:
A pull request have been merged to rtabmap repo to suppot both ZED sdk versions, this should make it the compilation easier with the different SDK versions: https://github.com/introlab/rtabmap/pull/501

bj-neilson, 2020-01-31:
Thanks @matlabbe  Matthieu 😁👍