stereolabs/zed-sdk #208: how to compile cpp within ROS

jinshengye-git, 2020-02-02:
Hi, I am using zed sdk v2.8.5  to make a  function to convert  sl::Mat   to   cv::Mat  in my own  ros pkg,  

` cv::Mat LineDetect::slMat2cvMat(sl::Mat& input) {
   int cv_type = -1;
   switch (input.getDataType()) {
       case sl::MAT_TYPE_F32_C1: cv_type = cv::CV_32FC1; break;
       case sl::MAT_TYPE_F32_C2: cv_type = cv::CV_32FC2; break;
       case sl::MAT_TYPE_F32_C3: cv_type = cv::CV_32FC3; break;
       case sl::MAT_TYPE_F32_C4: cv_type = cv::CV_32FC4; break;
       case sl::MAT_TYPE_U8_C1: cv_type = cv::CV_8UC1; break;
       case sl::MAT_TYPE_U8_C2: cv_type = cv::CV_8UC2; break;
       case sl::MAT_TYPE_U8_C3: cv_type = cv::CV_8UC3; break;
       case sl::MAT_TYPE_U8_C4: cv_type = cv::CV_8UC4; break;
       default: break;
   }
   return cv::Mat(input.getHeight(), input.getWidth(), cv_type, input.getPtr<sl::uchar     1>(sl::MEM_CPU));
}
`

when I `catkin_make` error comes like:

[image]

and I also tried to use g++ compile the single cpp file

$g++ -std=c++11 detect.cpp -I/usr/local/cuda-10.0/include -I/opt/ros/kinetic/include -I$HOME/catkin_ws/devel/include -lpthread  -lboost_system -lboost_date_time -lboost_thread -L/opt/ros/kinetic/lib  -lroscpp -lrosconsole -lrostime -lroscpp_serialization -lactionlib -Wl,-rpath,/opt/ros/kinetic/lib  -I/usr/local/zed/include -L/usr/local/zed/lib  \`pkg-config --cflags --libs opencv\` 

and the error is:

`
/tmp/cczG4XIu.o: In function `main':
detect.cpp:(.text+0x1943): undefined reference to `sl::Camera::open(sl::InitParameters)'
detect.cpp:(.text+0x197b): undefined reference to `sl::toString(sl::ERROR_CODE const&)'
detect.cpp:(.text+0x198a): undefined reference to `sl::String::c_str() const'
detect.cpp:(.text+0x19a1): undefined reference to `sl::String::~String()'
detect.cpp:(.text+0x19ab): undefined reference to `sl::Camera::close()'
detect.cpp:(.text+0x19c9): undefined reference to `sl::Camera::getResolution()'
detect.cpp:(.text+0x1a3e): undefined reference to `sl::Mat::Mat(unsigned long, unsigned long, sl::MAT_TYPE, sl::MEM)'
detect.cpp:(.text+0x1a5e): undefined reference to `LineDetect::slMat2cvMat(sl::Mat&)'
detect.cpp:(.text+0x1a8f): undefined reference to `LineDetect::obstacle_detectCallback(boost::shared_ptr<obstacle_msgs::sensors_<std::allocator<void> > const> const&)'
detect.cpp:(.text+0x1b66): undefined reference to `LineDetect::rmc_detectCallback(boost::shared_ptr<std_msgs::Bool_<std::allocator<void> > const> const&)'
detect.cpp:(.text+0x1c3d): undefined reference to `LineDetect::odometry_detectCallback(boost::shared_ptr<nav_msgs::Odometry_<std::allocator<void> > const> const&)'
detect.cpp:(.text+0x1d0f): undefined reference to `LineDetect::imu_detectCallback(boost::shared_ptr<sensor_msgs::Imu_<std::allocator<void> > const> const&)'
detect.cpp:(.text+0x1e2d): undefined reference to `LineDetect::roboline_operation_status(roboline::SrvTutorialRequest_<std::allocator<void> >&, roboline::SrvTutorialResponse_<std::allocator<void> >&)'
detect.cpp:(.text+0x1f33): undefined reference to `sl::Camera::grab(sl::RuntimeParameters)'
detect.cpp:(.text+0x2085): undefined reference to `LineDetect::cameraCombination(cv::Mat, cv::Mat, cv::Scalar_<double>, cv::Scalar_<double>, bool, cv::Scalar_<double>, cv::Scalar_<double>, double, std::array<double, 6ul>, bool, bool)'
detect.cpp:(.text+0x224c): undefined reference to `sl::Camera::retrieveImage(sl::Mat&, sl::VIEW, sl::MEM, int, int)'
detect.cpp:(.text+0x229e): undefined reference to `LineDetect::cameraOffsetCalibration(cv::Mat)'
detect.cpp:(.text+0x23e8): undefined reference to `LineDetect::colorthresh(cv::Mat, cv::Scalar_<double>, cv::Scalar_<double>, bool, cv::Scalar_<double>, cv::Scalar_<double>, double, std::array<double, 6ul>, bool, bool)'
detect.cpp:(.text+0x2438): undefined reference to `LineDetect::endOfLineMoveFwdCheck(int, bool, double)'
detect.cpp:(.text+0x247a): undefined reference to `LineDetect::xTurnCheck(int, bool, bool, int, int)'
detect.cpp:(.text+0x25a8): undefined reference to `sl::Mat::~Mat()'
detect.cpp:(.text+0x26d5): undefined reference to `sl::String::~String()'
detect.cpp:(.text+0x2938): undefined reference to `sl::Mat::~Mat()'
/tmp/cczG4XIu.o: In function `__static_initialization_and_destruction_0(int, int)':
detect.cpp:(.text+0x2aed): undefined reference to `sl::Camera::Camera()'
detect.cpp:(.text+0x2afc): undefined reference to `sl::Camera::~Camera()'
detect.cpp:(.text+0x2b0d): undefined reference to `sl::String::String()'
detect.cpp:(.text+0x2b19): undefined reference to `sl::InputType::InputType()'
detect.cpp:(.text+0x2b25): undefined reference to `sl::String::String()'
detect.cpp:(.text+0x2b34): undefined reference to `sl::String::String()'
detect.cpp:(.text+0x2ba3): undefined reference to `sl::String::~String()'
detect.cpp:(.text+0x2baf): undefined reference to `sl::String::~String()'
detect.cpp:(.text+0x2bc7): undefined reference to `sl::String::~String()'
detect.cpp:(.text+0x2dbc): undefined reference to `sl::String::~String()'
detect.cpp:(.text+0x2dcd): undefined reference to `sl::String::~String()'
/tmp/cczG4XIu.o:detect.cpp:(.text+0x2def): more undefined references to `sl::String::~String()' follow
/tmp/cczG4XIu.o: In function `sl::InitParameters::InitParameters(sl::RESOLUTION, int, int, sl::String, bool, sl::DEPTH_MODE, sl::UNIT, sl::COORDINATE_SYSTEM, bool, int, float, bool, bool, bool, int, sl::String, int, CUctx_st*, sl::InputType, sl::String, bool)':
detect.cpp:(.text._ZN2sl14InitParametersC2ENS_10RESOLUTIONEiiNS_6StringEbNS_10DEPTH_MODEENS_4UNITENS_17COORDINATE_SYSTEMEbifbbbiS2_iP8CUctx_stNS_9InputTypeES2_b[_ZN2sl14InitParametersC5ENS_10RESOLUTIONEiiNS_6StringEbNS_10DEPTH_MODEENS_4UNITENS_17COORDINATE_SYSTEMEbifbbbiS2_iP8CUctx_stNS_9InputTypeES2_b]+0xc5): undefined reference to `sl::String::String(sl::String const&)'
detect.cpp:(.text._ZN2sl14InitParametersC2ENS_10RESOLUTIONEiiNS_6StringEbNS_10DEPTH_MODEENS_4UNITENS_17COORDINATE_SYSTEMEbifbbbiS2_iP8CUctx_stNS_9InputTypeES2_b[_ZN2sl14InitParametersC5ENS_10RESOLUTIONEiiNS_6StringEbNS_10DEPTH_MODEENS_4UNITENS_17COORDINATE_SYSTEMEbifbbbiS2_iP8CUctx_stNS_9InputTypeES2_b]+0x132): undefined reference to `sl::String::String(sl::String const&)'
[... 2 lines cut ...]
detect.cpp:(.text._ZN2sl14InitParametersC2ENS_10RESOLUTIONEiiNS_6StringEbNS_10DEPTH_MODEENS_4UNITENS_17COORDINATE_SYSTEMEbifbbbiS2_iP8CUctx_stNS_9InputTypeES2_b[_ZN2sl14InitParametersC5ENS_10RESOLUTIONEiiNS_6StringEbNS_10DEPTH_MODEENS_4UNITENS_17COORDINATE_SYSTEMEbifbbbiS2_iP8CUctx_stNS_9InputTypeES2_b]+0x1a7): undefined reference to `sl::String::~String()'
detect.cpp:(.text._ZN2sl14InitParametersC2ENS_10RESOLUTIONEiiNS_6StringEbNS_10DEPTH_MODEENS_4UNITENS_17COORDINATE_SYSTEMEbifbbbiS2_iP8CUctx_stNS_9InputTypeES2_b[_ZN2sl14InitParametersC5ENS_10RESOLUTIONEiiNS_6StringEbNS_10DEPTH_MODEENS_4UNITENS_17COORDINATE_SYSTEMEbifbbbiS2_iP8CUctx_stNS_9InputTypeES2_b]+0x1bc): undefined reference to `sl::String::~String()'
/tmp/cczG4XIu.o: In function `sl::InitParameters::~InitParameters()':
detect.cpp:(.text._ZN2sl14InitParametersD2Ev[_ZN2sl14InitParametersD5Ev]+0x1a): undefined reference to `sl::String::~String()'
detect.cpp:(.text._ZN2sl14InitParametersD2Ev[_ZN2sl14InitParametersD5Ev]+0x3a): undefined reference to `sl::String::~String()'
detect.cpp:(.text._ZN2sl14InitParametersD2Ev[_ZN2sl14InitParametersD5Ev]+0x4a): undefined reference to `sl::String::~String()'
/tmp/cczG4XIu.o: In function `roboline::MyDirectionConfig::__get_statics__()':
detect.cpp:(.text._ZN8roboline17MyDirectionConfig15__get_statics__Ev[_ZN8roboline17MyDirectionConfig15__get_statics__Ev]+0x32): undefined reference to `dynamic_reconfigure::__init_mutex__'
/tmp/cczG4XIu.o: In function `sl::InitParameters::InitParameters(sl::InitParameters const&)':
detect.cpp:(.text._ZN2sl14InitParametersC2ERKS0_[_ZN2sl14InitParametersC5ERKS0_]+0xb4): undefined reference to `sl::String::String(sl::String const&)'
detect.cpp:(.text._ZN2sl14InitParametersC2ERKS0_[_ZN2sl14InitParametersC5ERKS0_]+0x145): undefined reference to `sl::String::String(sl::String const&)'
detect.cpp:(.text._ZN2sl14InitParametersC2ERKS0_[_ZN2sl14InitParametersC5ERKS0_]+0x170): undefined reference to `sl::InputType::InputType(sl::InputType const&)'
detect.cpp:(.text._ZN2sl14InitParametersC2ERKS0_[_ZN2sl14InitParametersC5ERKS0_]+0x190): undefined reference to `sl::String::String(sl::String const&)'
detect.cpp:(.text._ZN2sl14InitParametersC2ERKS0_[_ZN2sl14InitParametersC5ERKS0_]+0x1cf): undefined reference to `sl::String::~String()'
detect.cpp:(.text._ZN2sl14InitParametersC2ERKS0_[_ZN2sl14InitParametersC5ERKS0_]+0x1e4): undefined reference to `sl::String::~String()'
collect2: error: ld returned 1 exit status

`

my question : 

1. how to  config   CMakeLists.txt   within   ROS  catkin_ws/src ?
2. how to use g++ compile a single cpp code ?

Cheers

jinshengye-git, 2020-03-03:
by properly setting the CMakeLists.txt,  the issue was solved.

closed 2020-03-03 by jinshengye-git

ismail-JB, 2021-04-15:
hello, am working with zed and i have the same error that u had but am trying to compile the code within QT  without using cmake, what should i do to fix the problem ?

jsemeda, 2022-01-07:
Me too ... whatever happened to getResolution()? I am getting this error:
error C2039: 'getResolution': is not a member of 'sl::Camera'

jsemeda, 2022-01-07:
And where is sl_core64.lib?

Myzhar, 2022-01-07:
@ismail-JB @jsemeda This is a closed issue. Can you please open new issues by adding your questions with more information?
I suggest you write a post on the [Stereolabs community forum](https://community.stereolabs.com/) and use Github issues only for bug reporting and feature requests.
Thank you for understanding.