stereolabs/zed-sdk #209: Problem with getStreamingDeviceList

RamadanAhmed, 2020-02-02:
Hi,
I want to get all streaming devices but when I use this function, it returns 0
So what the problem ?
https://github.com/stereolabs/zed-examples/blob/57301b6560be99e55a746d1251d51ca708d4628e/camera%20streaming/receiver/src/main.cpp#L79

obraun-sl, 2020-02-03:
Hi,
Is your sender device on the same local network than the receiver ?

RamadanAhmed, 2020-02-03:
yes they both on the same switch and same working group(on windows)

obraun-sl, 2020-02-05:
OK.
Basically, getStreamingDeviceList() will look at the broadcast message send from each sender on port 20001. Once a message is caught with the informations filled, it will fill the vector.
Therefore make sure : 
- That the sender is launched and is running before launching the receiver
- port 20001 must be open on sender and receiver side (if not all port are available by default).

On windows, make sure that the applications are authorized for the network (popup that should appear first time you launch your program).

closed 2020-05-28 by RamadanAhmed