stereolabs/zed-sdk #210: ZED stream receiver : GLViewer.hpp error

PrivHateVoid, 2020-02-05:
Hello,

I just get my new ZED 2 camera today, and everything look to work great but i got one problem with the stream receiver code. 

When i try to run the code i got this error in VS2019 : (after translation)

> Can't open include file :'GLViewer.hpp' : No such file or directory.

I got no errors on my CMake results and this is the only project where i got this error.

The GLViewer.hpp file are on my include folder.

Best regard's, Ludo.

[image]

PrivHateVoid, 2020-02-06:
I have add the missing file into : 

C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\include

And now i have same error but with 'GL/glew.h', i'm C# programmer, i do not use C++ usually.

obraun-sl, 2020-02-06:
Hi, 
It seems there is an issue on this sample, since it does not require GLViewer.h/.cpp anymore but the file is still in the repository.
Try to delete the cmake cache (or just remove your build folder), then remove GLViewer.h/.cpp from the include/src directory and relaunch cmake again.
It should not try to include GLViewer anymore

PrivHateVoid, 2020-02-06:
Thank you.

Can you explain me about the stream protocol the ZED cam use?

I can't read the stream with my C# app, or VLC for exemple.

Can you write me the URI i can use for get the stream?

rtp://XXX.XXX.XXX.XXX:XXXX
http://XXX.XXX.XXX.XXX:XXXX
udp://XXX.XXX.XXX.XXX:XXXX?pkt_size=1316

Thank you inadvance.

obraun-sl, 2020-02-06:
The streaming module is made to be only received by the  ZED SDK, therefore it's not a standard stream you can read by VLC or other apps. 
Either use the ZED SDK in your app (we have a unity (C#) plugin for the ZED SDK that might help), or if you want to have a standard stream and not use the ZED SDK on the receiver part, then I suggest to look at gstreamer (sender and receiver) to have standard streaming.

PrivHateVoid, 2020-02-06:
ok thank you for that, its ok for me to use the ZED SDK, is just i'm new (and bad) in C++ and i have lot of openGL error when i try to compil and execute receive visual studio solution.

PS: 12 years later, i now remember my choice for C# at school.

Thank you for the links too.

Ludo

PrivHateVoid, 2020-02-06:
PS: It is possible from the ZED receiver code, to send the SbS video directly to HDMI output (without windows GUI) for plug a 3D TV?

closed 2020-02-10 by PrivHateVoid