stereolabs/zed-sdk #211: ZED_SVO_Export to sequence stuck at 99 percent

gaopinghuang0, 2020-02-07:
The .svo file is recorded via ZED-unity (plugin v2.8.2). Then I use `ZED_SVO_Export.exe <filename>.svo output_folder\ 2` to convert into sequence of images. It will get stuck at 99%, as shown in the screenshot. However, the images are already generated in the output_folder.

ZED SDK version: 2.8.5
Windows 10

[image]

kalyan0510, 2021-09-14:
+1 
I am facing very similar issue on Ubuntu 18.04
with 3.5-gl-devel-cuda10.0-ubuntu18.04 docker env.

obraun-sl, 2021-09-15:
Can you check here : 
https://github.com/stereolabs/zed-examples/issues/379
Make sure you are pulling the last version

closed 2022-04-27 by github-actions[bot]