stereolabs/zed-sdk #212: warning: libnvcuvid.so.1, needed by /usr/local/zed/lib/libsl_zed.so

isaacplayground, 2020-02-07:
I am trying to build the zed examples inside of a docker:
```
FROM stereolabs/zed:3.0-gl-devel-cuda10.2-ubuntu18.04

RUN apt-get update && apt-get install -y build-essential cmake git

RUN git clone https://github.com/stereolabs/zed-examples.git
WORKDIR 'zed-examples/svo recording/recording'
RUN cmake -DCMAKE_LIBRARY_PATH=/usr/local/cuda/lib64/stubs .
RUN make

CMD ["./ZED_SVO_Recording" "/store/zed_capture.svo"]
```

But the make command is failing with:
```
[ 50%] Building CXX object CMakeFiles/ZED_SVO_Recording.dir/src/main.o
[100%] Linking CXX executable ZED_SVO_Recording
/usr/bin/ld: warning: libnvcuvid.so.1, needed by /usr/local/zed/lib/libsl_zed.so, not found (try using -rpath or -rpath-link)
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cuvidDestroyDecoder'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cuvidUnmapVideoFrame64'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cuvidDestroyVideoParser'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cuvidMapVideoFrame64'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cuvidCtxLockCreate'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cuvidGetDecoderCaps'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cuvidParseVideoData'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cuvidDecodePicture'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cuvidCreateDecoder'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cuvidCtxLockDestroy'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cuvidCreateVideoParser'
collect2: error: ld returned 1 exit status
CMakeFiles/ZED_SVO_Recording.dir/build.make:99: recipe for target 'ZED_SVO_Recording' failed
make[2]: *** [ZED_SVO_Recording] Error 1
make[1]: *** [CMakeFiles/ZED_SVO_Recording.dir/all] Error 2
```
If I am understanding the problem correctly, libnvcuvid.so.1 isn't in the cuda docker as it is part of the nvidia driver, which is loaded at runtime from the host.

I have been trying to find a work around, but thus far I can't find a solution that isn't a complete hack. Help?

adujardin, 2020-02-10:
A far as I know there's no stub provided for libnvcuvid unfortunately.
This library is shared from the host when *running* a container (using nvidia runtime `--gpu ...`).

However, after looking into this issue in the nvidia docker repo, it seems there a workaround ([see here](https://github.com/NVIDIA/nvidia-docker/issues/775#issuecomment-400178344)) by using `'-Wl,--allow-shlib-undefined'`.

Replace the cmake command by : 

`RUN cmake -DCMAKE_LIBRARY_PATH=/usr/local/cuda/lib64/stubs -D CMAKE_CXX_FLAGS="-Wl,--allow-shlib-undefined" .`

closed 2022-04-27 by github-actions[bot]

ibrahimhroob, 2024-07-01:
We have faced the same issue, and we have figure out that `--allow-shlib-undefined` will just hide the issue and will not fix it. However we were able to fix it by enabling all nvidia driver capabilities in the docker file using this line `ENV NVIDIA_DRIVER_CAPABILITIES=all`, for more information you may refer here: [driver-capabilities](https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/1.10.0/user-guide.html#driver-capabilities)