stereolabs/zed-sdk #213: c++ code error LNK2019

PrivHateVoid, 2020-02-11:
Hello,

Please as C# developper i have somes difficulties with my ZED project.

I can run all samples and examples like a charm, but when i try to create my own project some weird errors when i try to compil my project, like you can see on the screen capture.

Any idea about that kind of error, can't found anything on google.

Thank you in advance.

[image]

adujardin, 2020-02-11:
Hi,

This error which can be translated by 'undefined reference' means the linker is not able to find some symbol (functions implementation), typically from an external library. In that case, seeing the namespace "cv::" indicates that it's functions from OpenCV that are missing.

The ZED SDK install a version of OpenCV and set the environment variable so that the cmake can find the configuration file of OpenCV, that's [this line for instance](https://github.com/stereolabs/zed-examples/blob/57301b6560be99e55a746d1251d51ca708d4628e/camera%20control/CMakeLists.txt#L19).

Make sure your cmake includes OpenCV if it's used in your project (it needs the find_package, [include_directories](https://github.com/stereolabs/zed-examples/blob/57301b6560be99e55a746d1251d51ca708d4628e/camera%20control/CMakeLists.txt#L24), [link_directories](https://github.com/stereolabs/zed-examples/blob/57301b6560be99e55a746d1251d51ca708d4628e/camera%20control/CMakeLists.txt#L28) and [target_link](https://github.com/stereolabs/zed-examples/blob/57301b6560be99e55a746d1251d51ca708d4628e/camera%20control/CMakeLists.txt#L42)).

PrivHateVoid, 2020-02-11:
Thank you very much for your fast anwser i will try ASAP.

i was not thinking about that because i have added all include in my project properties in VC++ include directories.

Anyway, thank you and i will be back to close the issus soon if that work.

PrivHateVoid, 2020-02-11:
Works perfectly, hard to learn but step by step i take pleasure to code on that little zed cam !!

Thank you !

PrivHateVoid, 2020-02-11:
3

closed 2020-02-11 by PrivHateVoid