stereolabs/zed-sdk #215: New SDK can't build the sample.

machenxiang, 2020-02-17:
My computer system is Ubuntu 16.04 and CUDA is 10.2. 
1.  I have succeeded install SDK(ZED_SDK_Ubuntu16_cuda10.2_v3.0.2.run). I only can open ZED Diagnostic Toll and ZED Exploer, but I can't open the others. And this is the result of ZED Diagnostic.

[image]

2. When I compile the sample in SDK, it occurs some errors.
When i cmake .. it occurs this
mcx@mcx-Lenovo-Rescuer-15ISK:/usr/local/zed/samples/camera control/build$ cmake ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at CMakeLists.txt:12 (cmake_policy):
  The OLD behavior for policy CMP0003 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:13 (cmake_policy):
  The OLD behavior for policy CMP0015 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Looking for sgemm_
-- Looking for sgemm_ - not found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Looking for sgemm_
-- Looking for sgemm_ - found
-- Found BLAS: /usr/lib/libopenblas.so  
-- Found CUDA: /usr/local/cuda (found version "10.2") 
-- Found OpenCV: /opt/ros/kinetic (found version "3.3.1") 
CMake Warning (dev) at CMakeLists.txt:20 (find_package):
  Ignoring EXACT since no version is requested.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /usr/local/zed/samples/camera control/build

When i make ,it occurs this.
mcx@mcx-Lenovo-Rescuer-15ISK:/usr/local/zed/samples/camera control/build$ make
Scanning dependencies of target ZED_Camera_Control
[ 50%] Building CXX object CMakeFiles/ZED_Camera_Control.dir/src/main.o
[100%] Linking CXX executable ZED_Camera_Control
/usr/bin/ld: warning: libturbojpeg.so.0, needed by /usr/local/zed/lib/libsl_zed.so, not found (try using -rpath or -rpath-link)
/usr/local/zed/lib/libsl_zed.so：对‘tjInitCompress@TURBOJPEG_1.0’未定义的引用
/usr/local/zed/lib/libsl_zed.so：对‘tjFree@TURBOJPEG_1.2’未定义的引用
/usr/local/zed/lib/libsl_zed.so：对‘tjDestroy@TURBOJPEG_1.0’未定义的引用
/usr/local/zed/lib/libsl_zed.so：对‘tjDecompress2@TURBOJPEG_1.2’未定义的引用
/usr/local/zed/lib/libsl_zed.so：对‘tjAlloc@TURBOJPEG_1.2’未定义的引用
/usr/local/zed/lib/libsl_zed.so：对‘tjCompress2@TURBOJPEG_1.2’未定义的引用
/usr/local/zed/lib/libsl_zed.so：对‘tjInitDecompress@TURBOJPEG_1.0’未定义的引用
collect2: error: ld returned 1 exit status
CMakeFiles/ZED_Camera_Control.dir/build.make:133: recipe for target 'ZED_Camera_Control' failed
make[2]: *** [ZED_Camera_Control] Error 1
CMakeFiles/Makefile2:75: recipe for target 'CMakeFiles/ZED_Camera_Control.dir/all' failed
make[1]: *** [CMakeFiles/ZED_Camera_Control.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

obraun-sl, 2020-02-17:
It seems that turbojpeg is missing. To install it, just type in a terminal : 
`$ sudo apt-get install libjpeg-turbo8-dev libturbojpeg`

Should fix it.

machenxiang, 2020-02-17:
> It seems that turbojpeg is missing. To install it, just type in a terminal :
> `$ sudo apt-get install libjpeg-turbo8-dev libturbojpeg`
> 
> Should fix it.

Thank you very much. I have fixed it by your ways.

closed 2020-02-21 by adujardin