stereolabs/zed-sdk #217: Edge detection availability in SDK?

xen20, 2020-02-17:
Hello,
I have been working with the Zed 2 camera for a few days now, and having done a few captures, the quality of the point cloud is quite good, however I would like to ask if any edge detection filters are available: something similar to this, taken with an Intel D435:

[image]

Which is an edge-filtered depth map of the following image:

[image]

With the D435, that output is achieved via very specific capture parameters - a lot more are tunable on it than the Zed 2 allows. Also, investigation of the SDK found no such functionality, is it planned or did I just miss it? :) 

thank you.

MohamedAliRashad, 2020-02-20:
I have a similar problem with the accuracy of the Depth Maps from the ZED Mini (in "FILL" Mode).

Until a formal solution comes from stereolabs, i think using an [edge kernel](https://stackoverflow.com/questions/4993082/how-to-sharpen-an-image-in-opencv) will be a good idea
```
kernel = np.array([[-1,-1,-1], [-1,9,-1], [-1,-1,-1]])
im = cv2.filter2D(im, -1, kernel)
```

P-yver, 2020-04-02:
Hi,
To filter the depth map you can use two different parameters:
`sl::RuntimeParameters::confidence_threshold`  
and/or 
`sl::RuntimeParameters::texture_confidence_threshold`
See: https://www.stereolabs.com/docs/api/structsl_1_1RuntimeParameters.html

BunnyIcon, 2020-11-18:
Hello, I'm facing the same problem, did you solve it? There are still some zeros in my depth array.

closed 2022-04-27 by github-actions[bot]