stereolabs/zed-sdk #224: SDK 3.0.3: Enable/disable recording fails after first recording

JesperChristensen89, 2020-02-27:
In the newest update, the recording issue when streaming is fixed. However, if you wish to split your recording in multiple files it fails after having recorded the first file. 

Example:

enable recording on file1.svo and record say 100 frames, then disable recording, change `rec_params.video_filename` and enable recording again. This new file will only become 280 bytes large, while the initial file records just fine.

obraun-sl, 2020-03-04:
Hi,

Issue has been identified and will be fixed in next release (3.1)

Best,
OB

prelee, 2020-03-09:
I don't know if it helps, I ran into this problem as well, a temporary solution (until 3.1) that is pseudo working for me is to actually close the camera and re-open it, then you can save multiple files, however it takes time and you lose recording for the time in between closing and re-opening.

JesperChristensen89, 2020-03-09:
@obraun-sl Any news as to when you will release the next version?

obraun-sl, 2020-03-10:
The SDK 3.1.0 (with the fix) has been released : 
https://www.stereolabs.com/developers/release/

closed 2022-04-27 by github-actions[bot]