stereolabs/zed-sdk #226: How to get the camera matrix for projections?

thsant, 2020-03-10:
[I know this is not an issue, but I could not find a proper forum to discuss this.]

I'm trying to get the _camera matrix_ P for the left frame (I would like to project 3-D points from  _fused point cloud_ to the _left_ camera frame). This should be given by P = K[R | t], being K the camera intrinsic matrix, R the camera rotation and t = -RC where C is the camera center in the world coordinates. 

In Python, I'm using:

```
K = np.array([[cam_params.left_cam.fx,                      0, cam_params.left_cam.cx],
              [                     0, cam_params.left_cam.fy, cam_params.left_cam.cy],
              [                     0,                      0,                      1]])

R = pose.get_rotation_matrix(sl.Rotation()).r     
t = pose.get_translation(sl.Translation()).get()            
world2cam = np.hstack((R, np.dot(-R, t).reshape(3,-1)))
P = np.dot(K, world2cam) 
```
In my plots, I can see that the cloud and the camera centers (computed from P) look OK, but the rotation looks weird (the principal ray is not pointing to the expected direction). Also the principal ray is varying according the `sl.COORDINATE_SYSTEM` setting. 

So, I would like to ask:

* Is an easier way there to project 3-D points from FusedPointCloud to the left frame?
* What's the proper way to extract the camera matrix from ZED?

thsant, 2020-03-12:
It took me a long time to discover, but we need to _transpose_ the rotation matrix given by the `get_rotation_matrix`:

```
R = pose.get_rotation_matrix(sl.Rotation()).r.T     
t = pose.get_translation(sl.Translation()).get()            
world2cam = np.hstack((R, np.dot(-R, t).reshape(3,-1)))
P = np.dot(K, world2cam)
```

closed 2020-03-12 by thsant

Huynh-Anh, 2021-10-06:
hI @thsant,
what is the pose object ?

kgnandanwar, 2023-03-22:
hI @thsant,
what is the pose object ?